ros-controls/ros2_control #596: executable 'spawner.py' not found on the libexec directory controller_manager

dsauval, 2021-12-17:
**Describe the bug**
When trying to use spawner.py using a python launchfile, I get the following error. I don't get this error using ros2 run :
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at C:\dev\ros2_foxy\Lib\site-packages\launch\launch_service.py:226> exception=SubstitutionFailure("executable 'spawner.py' not found on the libexec directory 'C:\Users\dsauval\webots_ws\install\controller_manager\lib\controller_manager' ")>

**To Reproduce**
Steps to reproduce the behavior (I tried on both the foxy and the galactic branches):
1. git clone https://github.com/ros-controls/ros2_control
2. In ros2_control folder : git checkout foxy
3. git clone https://github.com/ros-controls/control_msgs
4. In control_msgs folder : git checkout foxy-devel
5. Build the workspace with colcon build --event-handlers console_direct+
6. Launch a spwaner.py node from a launchfile, see the error.

Launchfile I used :

`from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    return LaunchDescription([
        Node(
            package='controller_manager',
            executable='spawner.py',
            output='screen',
            prefix='python.exe',
        )
    ])`

**Expected behavior**
I would expect to get the same output as if I would use ros2 run to start the node.

**Environment :**
 - OS: Windows 10 Pro, 2004, 
 - Version : Foxy 2021/10/13

bmagyar, 2021-12-20:
Checking the Foxy version I can't see anything that'd be obviously wrong:
* the file exists (we deprecated it in later versions but not in Foxy)
* there is an install statement which seems to point to the right spot (otherwise others would have complained too)

That unfortunately as far as I could go with this having no way to reproduce it locally.

Seeing the workspace name, are you following a recent webots tutorial by any chance? Have you tried asking if they had similar issues?

dsauval, 2021-12-20:
Hi, yes I asked them for help (https://github.com/cyberbotics/webots_ros2/issues/350).
They could successfully run everything using the same instructions I followed. I tried on another computer and it worked, meaning there is something wrong with my work machine. I will investigate and update here with the solution ;) 
I wonder if it can be related to the fact I use VS2017.

closed 2021-12-20 by dsauval