ros-controls/ros2_control #599: Port TransmissionLoader from ROS Noetic over to ROS2

bmagyar, 2021-12-26:
**Is your feature request related to a problem? Please describe.**
Currently, transmissions can only be used when one instantiates them by hand in the hardware component. In ROS1, one could add the requested transmissions to the xml and their parameters and have the transmission loader automatically take care of creating the right instances and connecting them to where they have to. 

**Describe the solution you'd like**
`TransmissionInfo` can be found [here](https://github.com/ros-controls/ros2_control/blob/master/hardware_interface/include/hardware_interface/hardware_info.hpp#L91).

For examples of configuring transmissions, look for `<transmission` in [this](https://github.com/ros-controls/ros2_control/blob/master/ros2_control_test_assets/include/ros2_control_test_assets/components_urdfs.hpp) file.

bmagyar, 2021-12-26:
Ping @VX792  please acknowledge you saw this ;)

VX792, 2021-12-26:
I did! :D

bmagyar, 2021-12-27:
Thanks, with the response I could assign the issue to you ;)

wmmc88, 2022-02-16:
#633 only resolves simple transmissions, right? The rest of this issue isn't resolved yet.

bmagyar, 2022-02-17:
That is correct, I'll tick off the first item on the list ;)