ros-controls/ros2_control #604: Async controllers

bmagyar, 2022-01-03:


FranekStark, 2024-10-30:
Hi, may I ask what is the status on that?
For many hierarchical controllers, which involve an optimization problem being solved at a lower frequency, this feature would be really needed.

saikishor, 2024-10-30:
> Hi, may I ask what is the status on that? For many hierarchical controllers, which involve an optimization problem being solved at a lower frequency, this feature would be really needed.

Hello @FranekStark!

you already have the async controllers implemented. We have a new implementation recently merged and released to rolling and jazzy, and we also have an old implementation. You can have it by simply setting the `is_async` parameter to true for the controller and remember to set the appropriate controller update rate needed for your optimization framework

saikishor, 2024-10-30:
https://github.com/ros-controls/ros2_control/pull/1489 should solve your issue :)

closed 2024-10-30 by saikishor

FranekStark, 2024-10-30:
@saikishor Thanks for you fast clarification! The documentation is a bit sparse on that, so I was a bit confused. :)

christophfroehlich, 2024-10-30:
@FranekStark Contributions to control.ros.org are very welcome, what would you like to add to make that clear? And where (ros2_control, ros2_controllers, a new demo?)

FranekStark, 2024-10-31:
@christophfroehlich I guess under ros2_control/concepts,  a document that explains how the scheduling of the controller updates works, would help a lot.

commit c5888791 (2025-07-15): Fix controller activation crash on macOS (Fixes #604) (#2391)
commit 865a596f (2025-07-16): Fix controller activation crash on macOS (Fixes #604) (#2391) (#2396)

(cherry picked from commit c588879194dfcce95bf25d4d7bb4a6691b456cb0)