ros-controls/ros2_control #606: Build fails due to use of abbreviated function template (C++20 feature)

wmmc88, 2022-01-06:
**Describe the bug**
#579 seems to have hadded an [auto used for the `target_interfaces` parameter](https://github.com/ros-controls/ros2_control/blob/8345f28dea6f993283f883b08f7f19ecde96b9d6/hardware_interface/include/fake_components/generic_system.hpp#L102) which seems to be [a feature of C++20 and onwards](https://en.cppreference.com/w/cpp/language/function_template#Abbreviated_function_template) and according to [REP2000](https://www.ros.org/reps/rep-2000.html), even ros2 Humble is still targeted to C++17.

I found this because master now fails to builds on windows after #579.

**Windows Build Failure**
```
[Processing: hardware_interface]
--- output: hardware_interface
Not searching for unused variables given on the command line.
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.22523.
-- The C compiler identification is MSVC 19.29.30138.0
-- The CXX compiler identification is MSVC 19.29.30138.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 1.2.0 (C:/dev/ros2_rolling/install/share/ament_cmake/cmake)
-- Found Python3: C:/python38/python.exe (found version "3.8.10") found components: Interpreter
-- Found control_msgs: 3.0.0 (C:/Users/melvi/git-repos/ros2_upstream_ws/install/share/control_msgs/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Found rosidl_adapter: 3.0.0 (C:/dev/ros2_rolling/install/share/rosidl_adapter/cmake)
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found pluginlib: 5.0.0 (C:/dev/ros2_rolling/install/share/pluginlib/cmake)
-- Found TinyXML2 via Config file: C:/ProgramData/chocolatey/lib/tinyxml2/share/cmake
-- Found rclcpp_lifecycle: 14.0.0 (C:/dev/ros2_rolling/install/share/rclcpp_lifecycle/cmake)
-- Found rmw_implementation_cmake: 5.1.0 (C:/dev/ros2_rolling/install/share/rmw_implementation_cmake/cmake)
-- Found rmw_cyclonedds_cpp: 1.1.1 (C:/dev/ros2_rolling/install/share/rmw_cyclonedds_cpp/cmake)
-- Using RMW implementation 'rmw_cyclonedds_cpp' as default
-- Found TinyXML2 via Config file: C:/ProgramData/chocolatey/lib/tinyxml2/share/cmake
-- Found ament_lint_auto: 0.11.3 (C:/dev/ros2_rolling/install/share/ament_lint_auto/cmake)
-- Found gmock sources under 'C:/dev/ros2_rolling/install/src/gmock_vendor': C++ tests using 'Google Mock' will be built
-- Found ament_cmake_gtest: 1.2.0 (C:/dev/ros2_rolling/install/share/ament_cmake_gtest/cmake)
-- Found gtest sources under 'C:/dev/ros2_rolling/install/src/gtest_vendor': C++ tests using 'Google Test' will be built
-- Added test 'copyright' to check source files copyright and LICENSE
-- Added test 'cppcheck' to perform static code analysis on C / C++ code
-- Configured cppcheck include dirs: C:/Users/melvi/git-repos/ros2_upstream_ws/src/ros-controls/ros2_control/hardware_interface/include
-- Configured cppcheck exclude dirs and/or files:
-- Added test 'lint_cmake' to check CMake code style
-- Added test 'xmllint' to check XML markup files
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/melvi/git-repos/ros2_upstream_ws/build/hardware_interface
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Checking Build System
  Building Custom Rule C:/Users/melvi/git-repos/ros2_upstream_ws/src/ros-controls/ros2_control/hardware_interface/CMakeLists.txt
  actuator.cpp
  component_parser.cpp
  resource_manager.cpp
  sensor.cpp
  system.cpp
C:\Users\melvi\git-repos\ros2_upstream_ws\src\ros-controls\ros2_control\hardware_interface\include\hardware_interface/resource_manager.hpp(294,41): warning C4251: 'hardware_interface::ResourceManager::claimed_command_interface_map_': class 'std::unordered_map<std::string,bool,std::hash<std::string>,std::equal_to<std::string>,std::allocator<std::pair<const std::string,bool>>>' needs to have dll-interface to be used by clients of class 'hardware_interface::ResourceManager' (compiling source file C:\Users\melvi\git-repos\ros2_upstream_ws\src\ros-controls\ros2_control\hardware_interface\src\resource_manager.cpp) [C:\Users\melvi\git-repos\ros2_upstream_ws\build\hardware_interface\hardware_interface.vcxproj]
C:\Users\melvi\git-repos\ros2_upstream_ws\src\ros-controls\ros2_control\hardware_interface\include\hardware_interface/resource_manager.hpp(294): message : see declaration of 'std::unordered_map<std::string,bool,std::hash<std::string>,std::equal_to<std::string>,std::allocator<std::pair<const std::string,bool>>>' (compiling source file C:\Users\melvi\git-repos\ros2_upstream_ws\src\ros-controls\ros2_control\hardware_interface\src\resource_manager.cpp) [C:\Users\melvi\git-repos\ros2_upstream_ws\build\hardware_interface\hardware_interface.vcxproj]
C:\Users\melvi\git-repos\ros2_upstream_ws\src\ros-controls\ros2_control\hardware_interface\include\hardware_interface/resource_manager.hpp(296,32): warning C4251: 'hardware_interface::ResourceManager::resource_lock_': class 'std::recursive_mutex' needs to have dll-interface to be used by clients of class 'hardware_interface::ResourceManager' (compiling source file C:\Users\melvi\git-repos\ros2_upstream_ws\src\ros-controls\ros2_control\hardware_interface\src\resource_manager.cpp) [C:\Users\melvi\git-repos\ros2_upstream_ws\build\hardware_interface\hardware_interface.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\mutex(96): message : see declaration of 'std::recursive_mutex' (compiling source file C:\Users\melvi\git-repos\ros2_upstream_ws\src\ros-controls\ros2_control\hardware_interface\src\resource_manager.cpp) [C:\Users\melvi\git-repos\ros2_upstream_ws\build\hardware_interface\hardware_interface.vcxproj]
C:\Users\melvi\git-repos\ros2_upstream_ws\src\ros-controls\ros2_control\hardware_interface\include\hardware_interface/resource_manager.hpp(297,36): warning C4251: 'hardware_interface::ResourceManager::resource_storage_': class 'std::unique_ptr<hardware_interface::ResourceStorage,std::default_delete<hardware_interface::ResourceStorage>>' needs to have dll-interface to be used by clients of class 'hardware_interface::ResourceManager' (compiling source file C:\Users\melvi\git-repos\ros2_upstream_ws\src\ros-controls\ros2_control\hardware_interface\src\resource_manager.cpp) [C:\Users\melvi\git-repos\ros2_upstream_ws\build\hardware_interface\hardware_interface.vcxproj]
C:\Users\melvi\git-repos\ros2_upstream_ws\src\ros-controls\ros2_control\hardware_interface\include\hardware_interface/resource_manager.hpp(297): message : see declaration of 'std::unique_ptr<hardware_interface::ResourceStorage,std::default_delete<hardware_interface::ResourceStorage>>' (compiling source file C:\Users\melvi\git-repos\ros2_upstream_ws\src\ros-controls\ros2_control\hardware_interface\src\resource_manager.cpp) [C:\Users\melvi\git-repos\ros2_upstream_ws\build\hardware_interface\hardware_interface.vcxproj]
C:\Users\melvi\git-repos\ros2_upstream_ws\src\ros-controls\ros2_control\hardware_interface\src\component_parser.cpp(341,91): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data [C:\Users\melvi\git-repos\ros2_upstream_ws\build\hardware_interface\hardware_interface.vcxproj]
C:\Users\melvi\git-repos\ros2_upstream_ws\src\ros-controls\ros2_control\hardware_interface\src\component_parser.cpp(351,87): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data [C:\Users\melvi\git-repos\ros2_upstream_ws\build\hardware_interface\hardware_interface.vcxproj]
     Creating library C:/Users/melvi/git-repos/ros2_upstream_ws/build/hardware_interface/Release/hardware_interface.lib and object C:/Users/melvi/git-repos/ros2_upstream_ws/build/hardware_interface/Release/hardware_interface.exp
[... cut ...]
C:\Users\melvi\git-repos\ros2_upstream_ws\src\ros-controls\ros2_control\hardware_interface\include\hardware_interface/resource_manager.hpp(297): message : see declaration of 'std::unique_ptr<hardware_interface::ResourceStorage,std::default_delete<hardware_interface::ResourceStorage>>' [C:\Users\melvi\git-repos\ros2_upstream_ws\build\hardware_interface\test_resource_manager.vcxproj]
  test_resource_manager.vcxproj -> C:\Users\melvi\git-repos\ros2_upstream_ws\build\hardware_interface\Release\test_resource_manager.exe
---
Failed   <<< hardware_interface [34.1s, exited with code 1]
```

**Environment (please complete the following information):**
 - OS: Windows
 - Version: master

wmmc88, 2022-01-06:
It also causes the following warning when building in CI:
```
  In file included from /root/target_ws/src/ros2_control/hardware_interface/src/fake_components/generic_system.cpp:17:
  /root/target_ws/src/ros2_control/hardware_interface/include/fake_components/generic_system.hpp:102:5: warning: use of ‘auto’ in parameter declaration only available with ‘-fconcepts’
    102 |     auto & target_interfaces, bool using_state_interfaces);
        |     ^~~~
  /root/target_ws/src/ros2_control/hardware_interface/src/fake_components/generic_system.cpp:464:3: warning: use of ‘auto’ in parameter declaration only available with ‘-fconcepts’
    464 |   auto & target_interfaces, bool using_state_interfaces)
        |   ^~~~
```

bmagyar, 2022-01-06:
@bailaC could you please address this? I think a template could be used instead of auto there.

bailaC, 2022-01-06:
sure, I will check this.

wmmc88, 2022-01-13:
fixed in #608

closed 2022-01-13 by wmmc88