ros-controls/ros2_control #609: rqt_controller_manager port

wmmc88, 2022-01-06:
Are there any plans to port [rqt_controller_manager](https://github.com/ros-controls/ros_control/tree/noetic-devel/rqt_controller_manager) from ros1 `ros_control`? Or is there something else that provides/replaces that functionality?

bmagyar, 2022-01-07:
There is nothing stopping anyone from porting it but we don't have it in the works at the moment. Are you interested in doing it?

wmmc88, 2022-01-07:
> There is nothing stopping anyone from porting it but we don't have it in the works at the moment. Are you interested in doing it?

I don't have the time to take it on at the moment, but if I get some time in the next couple of weeks, i'll throw up a pr :)

bmagyar, 2022-01-10:
Thank you for contributing to the roadmap, added!

bmagyar, 2022-11-05:
https://github.com/ros-controls/ros2_control/pull/813

PR #813 (2022-11-17): rqt controller manager ros 2 port
This is a port of rqt_controller_manager from ROS 1 to ROS 2, but with some modifications listed below that were necessary due to fundamental changes in the ros2 control architecture. This PR probably still requires some cleaning up, but I'd appreciate any early reviews and testers!

  - states of the controllers have changed
    - not loaded
    - unconfigured
    - inactive
    - active
    - finalized
  - the controller namespace has been replaced with controller node name
  - parameters are obtained directly from controller manager using get_parameter serv [... cut]
closed 2022-11-17 by bmagyar