ros-controls/ros2_control #612: [Question] Runtime (re)configuration of available controllers to the controller manager?

tlpss, 2022-01-10:
Hi, 

I'm in the following situation: I have a package with the description of my robot and a package with the description of my gripper. Both of which use ros2_controllers that are defined in their respective `controllers.yaml` file. 

Now I would like to bringup both the gripper and the robot together, but I would prefer not to duplicate the controller configurations or launch files into a new configuration for the combined robot + gripper. This would require to start up the manager in the first launch file and add the controllers of the gripper to that manager in the second launch file, I believe. 

Would this be possible or is my only option to manually create a new configuration file and a new launch file to start the manager with the combined set of controllers? 

Sorry if my terminology is not always correct, but I'm rather new to the control framework..

Kind regards 

Thomas

bmagyar, 2022-01-10:
Hi Thomas,

I believe this is currently not possible but I think it is a very reasonable use-case. Can you write up please how you'd like this to work, perhaps even as a demo, adding a gripper to the rrbot in ros2_control_demos on a PR? We can trace our steps back from there and make the framework meet the requirement ;)

tlpss, 2022-01-11:
Thanks for your reply. 

> can you write up please how you'd like this to work 

I'm not sure exactly, but here's how I see it:

a typical piece of hardware will have following 3 packages: 
- the driver pcckage, implementing a HW interface
- the description package, containing a description in URDF with the control tags 
- the bringup package that has a launch file and configuration of the relevant controllers

To combine the pieces, you need to combine the URDFs in a single URDF (a.o. required for Moveit! which does not allow dynamic reconfiguration of the robot_description if I'm not mistaken). 

Also, you need to create a single controllers.yaml file and a new launch file in which you start up the controller manager only once, pass the combined controllers configuration and start the controllers for both pieces of hardware. This is the part where a lot of duplication is going on. 

In an ideal scenario you could just run the launch files from both pieces of hardware separately (if required with a flag for the second launch file to tell that the controller manager is already running, bc it [seems](https://stackoverflow.com/questions/50954601/launch-node-if-not-already-running)  that this cannot be discovered within a launch file, and this would require a way to to update the controller configuration with a new set of controllers for the second piece of hardware from within the launch file. I'm not sure what would be the best way to achieve this.

bmagyar, 2022-01-11:
Thanks, this is exactly what I was looking for!

tlpss, 2022-01-11:
Okay great! 

I would be willing to provide some help in implementing this once it is decided how to integrate this.

destogl, 2022-01-12:
For now I think you have to only add a new launch file and then load both `controllers.yaml` file to it

tlpss, 2022-01-14:
> For now I think you have to only add a new launch file and then load both controllers.yaml file to it

that's true!  Even though it seems nice to me if you would not need to create a new launch file either