ros-controls/ros2_control #613: control manager crashes at bringup due to "undefined symbol" error.

tlpss, 2022-01-12:
**Describe the bug**
control manager crashes at startup due to "undefined symbol" error. Not sure if it is bug, could always be due to misconfigurations on my behalf..

**To Reproduce**
Steps to reproduce the behavior:
1. pull moveit/moveit2:rolling-release docker file
2. apt-get install ros2-control and ros2-controllers
3. clone the ros2_control_demos repo
4. colcon build 
4. run `ros2 launch ros2_control_demo_bringup rrbot.launch.py`

output: 
```
root@tlips-XPS:/workspaces/rolling_colcon_ws/ROS# source install/setup.bash 
root@tlips-XPS:/workspaces/rolling_colcon_ws/ROS# ros2 launch ros2_control_demo_bringup rrbot.launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2022-01-12-11-01-11-470200-tlips-XPS-22880
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [22884]
[INFO] [robot_state_publisher-2]: process started with pid [22886]
[INFO] [spawner-3]: process started with pid [22888]
[robot_state_publisher-2] Link base_link had 1 children
[robot_state_publisher-2] Link link1 had 1 children
[robot_state_publisher-2] Link link2 had 3 children
[robot_state_publisher-2] Link camera_link had 1 children
[robot_state_publisher-2] Link camera_link_optical had 0 children
[robot_state_publisher-2] Link hokuyo_link had 0 children
[robot_state_publisher-2] Link tool_link had 0 children
[robot_state_publisher-2] [INFO] [1641985271.900344672] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1641985271.900401840] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-2] [INFO] [1641985271.900411361] [robot_state_publisher]: got segment camera_link_optical
[robot_state_publisher-2] [INFO] [1641985271.900417485] [robot_state_publisher]: got segment hokuyo_link
[robot_state_publisher-2] [INFO] [1641985271.900423585] [robot_state_publisher]: got segment link1
[robot_state_publisher-2] [INFO] [1641985271.900429649] [robot_state_publisher]: got segment link2
[robot_state_publisher-2] [INFO] [1641985271.900435458] [robot_state_publisher]: got segment tool_link
[robot_state_publisher-2] [INFO] [1641985271.900441228] [robot_state_publisher]: got segment world
[ros2_control_node-1] [INFO] [1641985271.900509517] [RRBotSystemPositionOnlyHardware]: Configuring ...please wait...
[ros2_control_node-1] [INFO] [1641985271.900537665] [RRBotSystemPositionOnlyHardware]: Successfully configured!
[ros2_control_node-1] [INFO] [1641985271.900618740] [RRBotSystemPositionOnlyHardware]: Activating ...please wait...
[ros2_control_node-1] [INFO] [1641985271.900624294] [RRBotSystemPositionOnlyHardware]: Successfully activated!
[ros2_control_node-1] /opt/ros/rolling/lib/controller_manager/ros2_control_node: symbol lookup error: /opt/ros/rolling/lib/libcontroller_manager.so: undefined symbol: _ZN22rosidl_typesupport_cpp31get_service_type_support_handleIN23controller_manager_msgs3srv22ListHardwareComponentsEEEPK29rosidl_service_type_support_tv
[ERROR] [ros2_control_node-1]: process has died [pid 22884, exit code 127, cmd '/opt/ros/rolling/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_0k3egv9s --params-file /workspaces/rolling_colcon_ws/ROS/install/ros2_control_demo_bringup/share/ros2_control_demo_bringup/config/rrbot_controllers.yaml -r /forward_position_controller/commands:=/position_commands'].
[spawner-3] [INFO] [1641985272.024561711] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner-3] [INFO] [1641985274.051469290] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner-3] [INFO] [1641985276.078202630] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner-3] [INFO] [1641985278.109307055] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner-3] [INFO] [1641985280.142147117] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner-3] [ERROR] [1641985282.173278257] [spawner_joint_state_broadcaster]: Controller manager not available
[ERROR] [spawner-3]: process has died [pid 22888, exit code 1, cmd '/opt/ros/rolling/lib/controller_manager/spawner joint_state_broadcaster --controller-manager /controller_manager --ros-args'].
[INFO] [spawner-4]: process started with pid [23024]
[INFO] [rviz2-5]: process started with pid [23026]
[rviz2-5] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[spawner-4] [INFO] [1641985282.576908320] [spawner_forward_position_controller]: Waiting for /controller_manager services
[rviz2-5] [INFO] [1641985282.854329874] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-5] [INFO] [1641985282.854456300] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-5] [INFO] [1641985282.876838836] [rviz2]: Stereo is NOT SUPPORTED
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[rviz2-5] [INFO] [1641985282.941619702] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-2] [INFO] [1641985282.941616335] [rclcpp]: signal_handler(signum=2)
[spawner-4] Traceback (most recent call last):
[spawner-4]   File "/opt/ros/rolling/lib/controller_manager/spawner", line 20, in <module>
[spawner-4]     ret = main()
[spawner-4]   File "/opt/ros/rolling/lib/controller_manager/spawner.py", line 118, in main
[spawner-4]     if not wait_for_controller_manager(node, controller_manager_name,
[spawner-4]   File "/opt/ros/rolling/lib/controller_manager/spawner.py", line 60, in wait_for_controller_manager
[spawner-4]     time.sleep(0.2)
[spawner-4] KeyboardInterrupt
[rviz2-5] [ERROR] [1641985282.945655043] [rviz2]: Could not load display config: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at /tmp/binarydeb/ros-rolling-rcl-4.0.0/src/rcl/guard_condition.c:67
[INFO] [robot_state_publisher-2]: process has finished cleanly [pid 22886]
[INFO] [rviz2-5]: process has finished cleanly [pid 23026]
[ERROR] [spawner-4]: process has died [pid 23024, exit code -2, cmd '/opt/ros/rolling/lib/controller_manager/spawner forward_position_controller -c /controller_manager --ros-args'].
```
**Expected behavior**
ros2 controllers and manager starting up. This worked a few days ago in a similar setup.

**Additional information**
- version of the apt binaries: (2.0.0-1focal.20220105.200159)

**Environment (please complete the following information):**
 - OS: ubuntu 20.04
 - version: rolling
- inside docker container provided by Moveit

**Context**
Could not bring up the UR_ROS2_drivers after rebuilding my container this morning. Went back to check if the demos were still working but got the exact same error.

bmagyar, 2022-01-12:
Does *this* fix the issue? https://github.com/ros-controls/ros2_control/pull/580

bmagyar, 2022-01-12:
I'm afraid this isn't as simple as filing a bug report ;) 

You are using ROS Rolling, binary incompatibilities are expected there which is exactly what this seems to be. Please make sure to update *everything* from Rolling and do a clean build of the code you are building locally.

tlpss, 2022-01-12:
>  Please make sure to update everything

yeah should have done this more carefully, after upgrading *all* rolling binaries, everything works as expected.

Thanks for your quick reply!

closed 2022-01-12 by tlpss