ros-controls/ros2_control #622: FakeSystem initial_value does not seem to work correctly

AndyZe, 2022-01-20:
I am launching one of the common test packages from MoveIt, the Panda. The ros2_control xacro is here:

https://github.com/ros-planning/moveit_resources/blob/ros2/panda_moveit_config/config/panda.ros2_control.xacro

I saw this deprecation warning:

[ros2_control_node-5] [WARN] [1642614151.127346724] [fake_generic_system]: The usage of initial_position has been deprecated. Please use 'initial_value' instead.

So I tried updating `initial_position` to `initial_value` in that ros2_control xacro file. Yet, the robot does not move to the initial position like it used to.

AndyZe, 2022-01-20:
I wonder if it's because these joints have 2 state interfaces: position and velocity. Perhaps it's because this line should be the command interface, not the state interface?

`for (const auto & interface : joint.state_interfaces)`

AndyZe, 2022-01-20:
So can I ask how this feature is supposed to work? I don't see any usage example in ros2_control_demos.

Is it:

```
<state_interface name="position" initial_value="0.0"/>
<state_interface name="velocity" initial_value="0.0"/>
```

or

```
                <param name="initial_value">0.0</param>
                <command_interface name="position" />
                <state_interface name="position"/>
                <state_interface name="velocity"/>
```

I hope it's option #1. Option #2 wouldn't make sense if there are multiple state interfaces.

closed 2022-01-21 by AndyZe

wmmc88, 2022-01-22:
> @destogl can I ask how this feature is supposed to work? I don't see any usage example in ros2_control_demos.
> 
> Is it:
> 
> ```
> <state_interface name="position" initial_value="0.0"/>
> <state_interface name="velocity" initial_value="0.0"/>
> ```
> 
> or
> 
> ```
>                 <param name="initial_value">0.0</param>
>                 <command_interface name="position" />
>                 <state_interface name="position"/>
>                 <state_interface name="velocity"/>
> ```
> 
> I hope it's option 1. Option 2 wouldn't make sense if there are multiple state interfaces.

For anyone stumbling upon this looking for the new syntax, there's an example in the test file:

https://github.com/ros-controls/ros2_control/blob/a362e2bfbe2308291f20cfc0109c4e1fbf11fb50/hardware_interface/test/fake_components/test_generic_system.cpp#L259

AndyZe, 2022-01-22:
Thanks @wmmc88!