ros-controls/ros2_control #631: missing dependency in 2.1.0 when building on rolling-main

tylerjw, 2022-01-29:
Recently we added 2.1.0 of ros2_control to our repos file for upstream dependencies.  It is in the testing repo but not in the main repo yet.  When we try to build it in our CI it fails to build with this error:

```
  --- stderr: controller_manager_msgs
  make[2]: *** No rule to make target '/opt/ros/rolling/share/builtin_interfaces/cmake/../../../lib/libbuiltin_interfaces__python.so', needed by 'rosidl_generator_py/controller_manager_msgs/libcontroller_manager_msgs__python.so'.  Stop.
  make[1]: *** [CMakeFiles/Makefile2:487: CMakeFiles/controller_manager_msgs__python.dir/all] Error 2
  make[1]: *** Waiting for unfinished jobs....
  make: *** [Makefile:141: all] Error 2
  ---
  Failed   <<< controller_manager_msgs [0.93s, exited with code 2]
  Aborted  <<< hardware_interface [22.8s]
  
  Summary: 1 package finished [23.3s]
    1 package failed: controller_manager_msgs
```

I think this means that controller_manager_msgs is missing a dependency in the package.xml and/or in the CMakeLists.txt

Here is an example failed CI job: https://github.com/ros-planning/moveit2/runs/4992328626?check_suite_focus=true

bmagyar, 2022-01-29:
Are you 100% sure? I remember similar issues back when we started working
more actively with ROS2 and some dirty workspaces caused this. Any build
cache by any chance or ABI incompatibility?

tylerjw, 2022-01-29:
As ros2_control seems to be building fine on the buildfarm and is in the rolling-testing repo I am inclined to believe there is just something wrong with our CI.  I do get the warning in my CI about overriding ros2_control ros packages which seems really weird to me.  I'm going to re-build our docker image to see if that fixes it.  I'll close this if that fixes it.

tylerjw, 2022-01-30:
There is definitely something weird in our moveit2 CI with the docker image but I just experienced this same issue over on my fork of bio_ik which does not have a base docker image that it is using.  It just checks out all the upstream repos into a workspace and uses rosdep to install dependencies.

https://github.com/tylerjw/bio_ik/runs/4995798901?check_suite_focus=true

destogl, 2022-01-31:
It seems there is something wrong with [moveit_msgs package](https://github.com/tylerjw/bio_ik/runs/4995798901?check_suite_focus=true#step:7:1420). Or am I missing something?

tylerjw, 2022-01-31:
I don't know why I didn't see that.  It does seem like it is moveit_msgs that is causing the failure.  I had just assumed it had something to do with the warning that is being printed at the top:

```
  [0.418s] WARNING:colcon.colcon_core.verb:Some selected packages are already built in one or more underlay workspaces:
  	'hardware_interface' is in: /opt/ros/rolling
  	'controller_interface' is in: /opt/ros/rolling
  If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
  If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.
  
  If you understand the risks and want to override a package anyways, add the following to the command line:
  	--allow-overriding controller_interface hardware_interface
```

Thank you for pointing that out.  I'm going to see if I can reproduce the failure with moveit_msgs.  We also have a release of that that is half way synced (in testing but not main) and that might actually be what is failing here.  The warnings about  controller_interface hardware_interface still seem really weird.  At this point I think they might be caused by some weirdness with Docker or other cache like @bmagyar  you suggested before.  I'm sorry for posting this here.  I'll close this until I know more.

closed 2022-01-31 by tylerjw