ros-controls/ros2_control #632: Use ros2 control with sdf model

karimrahimi76, 2022-01-31:
hi , i am trying to use ros2 control for a parallel robot , and my model is sdf format.
when i want to launch a  project , controller manager needs robot description. but sdf format has no any robot tag.
even i tried to pass my project world and sdf format to controller manager, but it doesn't  work too.

would anybody mind Is ros2 control works with sdf format? 
 how can I handle it?

destogl, 2022-01-31:
Hi @karimrahimi76,

We are currently working with URDF file format, which is standard for ROS visualization/simulation/manipulation. Is there a reason you cannot use URDF format? How are you then visualizing your robot in `rviz`? To be sure to work with `ros2_control` try to define URDF format and give it to  the controller manager node. As a first step, you don't need any geometric description, but you can only provide `<ros2_control>` tag as shown in our [examples](https://github.com/ros-controls/ros2_control_demos/blob/master/README.md) (check the linked files for each example.

karimrahimi76, 2022-02-01:
> Hi @karimrahimi76,
>
> We are currently working with URDF file format, which is standard for ROS visualization/simulation/manipulation. Is there a reason you cannot use URDF format? How are you then [...]

thank you for your response , i cannot use urdf because i want to simulate parallel robots and need "ball" joint and "universal" joint that urdf doesn't support that.
so i can not simulate my robot in urdf format and i have to do it in sdf format.

olivier-stasse, 2022-02-01:
Dear @karimrahimi76,
I do not think that this is a software problem for ros2_control itself. 

In SDF it is needed because you *need* to simulate the closed kinematic chain. On the real robot, the mechanics is doing this for you. Strictly speaking you can write a URDF allowing to load the hardware interface for accessing the low level hardware of your robot. The URDF model will not include the close kinematic chain. So if your goal is to control your real robot with ros2_control there is no immediate problem which forbid you to do so.

The two main caveats are the following:
* As far as I know there is no ros2_controller able to take into account the parallel mechanism, but you can write your own.
* Using gazebo_ros2_package will be a challenge because it is using the URDF model to build the *simulated* model. A quick solution is to use the same URDF  than for the hardware_interface, but then the impact of this constraint will not appear. Right now I do not think it is directly feasible to use your SDF file instead (but it is probably doable).

Note: this remark is only a view of my own understanding of ros2_control.

destogl, 2022-02-01:
> thank you for your response , i cannot use urdf because i want to simulate parallel robots and need "ball" joint and "universal" joint that urdf doesn't support that. so i can not simulate my robot in urdf format and i have to do it in sdf format.

@karimrahimi76 as I wrote… you only need to provide <ros2_control> part of URDF which is kinematics-agnostic…

karimrahimi76, 2022-02-03:
> * Using gazebo_ros2_package will be a challenge because it is using the URDF model to build the _simulated_ model. A quick solution is to use the same URDF  th

dear @olivier-stasse  thank you for your attention and response.
I think you are fully right and we have too  write a custom package or library for our parallel Robot.

karimrahimi76, 2022-02-03:
> > thank you for your response , i cannot use urdf because i want to simulate parallel robots and need "ball" joint and "universal" joint that urdf doesn't support that. so i can not simulate my robot in urdf format and i have to do it in sdf format.
> 
> @karimrahimi76 as I wrote… you only need to provide <ros2_control> part of URDF which is kinematics-agnostic…
> 
> @olivier-stasse thanks for the clarification. I think your observations are fully right!

dear @destogl 
I checked the Examples and it was useful for me.
thank you for that. finally I think i have too write my custom package or plugin for my parallel Robot Close Chain Control.

destogl, 2022-02-14:
> dear @destogl I checked the Examples and it was useful for me. thank you for that. finally I think i have too write my custom package or plugin for my parallel Robot Close Chain Control.

The clue is that you ros2_control doesn't know anything about kinematics so you can also make it "broken" in URDF and things should work OK.

asimonov, 2024-09-06:
can you re-open this issue?

Gazebo Harmonic allows to use SDF for robot description, instead of URDF. mentioned as last item here: https://github.com/gazebosim/gz-harmonic/blob/main/highlights.md

but if I try to run this setup with ros2_control I see this:

```
[ros2_control_node-3] [INFO] [1725603736.262263586] [controller_manager]: Received robot description from topic.
[ros2_control_node-3] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-3]   what():  the robot tag is not root element in URDF
[ERROR] [ros2_control_node-3]: process has died
```

robot_state_publisher and joint_state_publisher work with SDF. would be nice to make ros2_control work with it too

christophfroehlich, 2024-09-09:
> Gazebo Harmonic allows to use SDF for robot description, instead of URDF. mentioned as last item here: https://github.com/gazebosim/gz-harmonic/blob/main/highlights.md, documentation is here https://gazebosim.org/docs/harmonic/ros2_interop/
> 
> but if I try to run this setup with ros2_control I see this:
> 
> ```
> [ros2_control_node-3] [INFO] [1725603736.262263586] [controller_manager]: Received robot description from topic.
> [ros2_control_node-3] terminate called after throwing an instance of 'std::runtime_error'
> [ros2_control_node-3]   what():  the robot tag is not root element in URDF
> [ERROR] [ros2_control_node-3]: process has died
> ```
> 
> robot_state_publisher and joint_state_publisher work with SDF. would be nice to make ros2_control work with it too

I haven't tried this out, but the description says that the example is using https://github.com/ros/sdformat_urdf to create a URDF from the SDF, which is then passed to the ROS 2 world.

azeey, 2024-09-24:
I believe `robot_state_publisher` works because it doesn't parse the URDF directly. Instead, it uses `urdf_parser_plugin` via `urdf::Model::InitString`
https://github.com/ros/robot_state_publisher/blob/65e4d964e66dab348aac43053855c584dbddd620/src/robot_state_publisher.cpp#L164. This loads the sdformat_urdf plugin which populates the `urdf::Model` based on the SDFormat file or string. It doesn't actually generate a URDF string. I believe, overall, this is the preferred approach since it's the most flexible and can support other file formats such as Collada
[image]

`joint_state_publisher` on the other hand is a python package, so it is not able to take advantage of that plugin mechanism. Instead, it handles URDF, SDF, and Collada separately based on the root tag of the input string.

https://github.com/ros/joint_state_publisher/blob/fb29caa6a341556ef8716b583ef32507d001dcdc/joint_state_publisher/joint_state_publisher/joint_state_publisher.py#L286-L291

I see ros2_control already uses `urdf_parser_plugin`: https://github.com/ros-controls/ros2_control/blob/84e85f9f0bbd67af9eebbe40bf5eef89792945ff/hardware_interface/src/component_parser.cpp#L838

But, I don't believe the urdfdom data structures keep extra data, such as the ros2_control tags, so it will probably have to handle SDF files separately. I think changing only a small section of code might get us there. 

If we also allowed for `<sdf>` tag in 
https://github.com/ros-controls/ros2_control/blob/84e85f9f0bbd67af9eebbe40bf5eef89792945ff/hardware_interface/src/component_parser.cpp#L818-L820

perhaps everything will "just work" 🤞🏾 .

PR #1763 (2024-11-07): Add Support for SDF
Fixes #632
I have modified `hardware_interface/src/component_parser.cpp` to add support for SDF models.

To test this out the demos being added through https://github.com/ros-controls/gz_ros2_control/pull/427 can be used.