ros-controls/ros2_control #644: Main control loop spin rate diverges with increasing CM update_rate

AndyZe, 2022-02-12:
I dropped a high-resolution timer in controller_manager/ros2_control_node.cpp (see attachment) then ran the rrbot demo with several different `update_rates`. Here are some results:

10 Hz update rate:       10.0084s for 100 iterations. Practically perfect.
1000 Hz update_rate:  10.5889s for 1000 iterations. Missed the `update_rate` by 5.9%

Conclusion:  the main loop spin rate diverges with increasing control rate.

[ros2_control_node.cpp.txt](https://github.com/ros-controls/ros2_control/files/8054357/ros2_control_node.cpp.txt)

AndyZe, 2022-02-12:
A contributing factor might be that a small bit of work is not accounted for while sleeping?

```
    while (rclcpp::ok())
    {
      rclcpp::Time begin_last = begin;  // THIS WORK IS NOT ACCOUNTED FOR WHILE SLEEPING
      begin = cm->now();
      cm->read();
      cm->update(begin, begin - begin_last);
      cm->write();
      rclcpp::Time end = cm->now();
      std::this_thread::sleep_for(std::max(
        std::chrono::nanoseconds(0),
        std::chrono::nanoseconds(1000000000 / cm->get_update_rate()) -
          std::chrono::nanoseconds(end.nanoseconds() - begin.nanoseconds())));
    }
```

Other than that, I suppose read(), update(), or write() just cannot keep up with the requested pace.

destogl, 2022-02-12:
@AndyZe it seems that you did something similar to my proposal in #645. Is this correct? If so, we can close that issue too with #646.

AndyZe, 2022-02-12:
Well, #646 doesn't completely fix it. There's still something else going on.

destogl, 2022-02-12:
I am not sure if `begin = now()` is placed properly...

AndyZe, 2022-02-15:
Agree, that's what #646 changes. It puts `begin = now()` right at the beginning of the iteration so it doesn't miss anything.

gavanderhoorn, 2022-02-16:
Originally in https://github.com/ros-controls/ros2_control/pull/647#issuecomment-1041246176:

> Has moving to a timer been discussed?
> 
> I'm not sure whether the current state of the executors is such that they can be used for reliable timing (I've seen [this note](https://github.com/ros-controls/ros2_control/blob/fdd08850a68e67c4bd70bad8d5b7cc55bf4fcbc4/controller_manager/src/ros2_control_node.cpp#L36-L39)), but there are other alternatives to setup timer callbacks (ie: non-ROS).
> 
> Using timers / callbacks would:
> 
>     1. be quite similar to how you'd do this in other control frameworks
> 
>     2. avoid cumulative drift
> 
>     3. if backed by a hw-timer: be potentially _very_ accurate
> 
>     4. be easier to read / grok
> 
>     5. allow detecting overruns easier (either the OS or the RTL typically has support for that)
> 
> 
> any type of `sleep(..)` always has the potential to drift in my experience.