ros-controls/ros2_control #645: [ros2_control_node] Let's check if we are measuring time correctly

destogl, 2022-02-12:
A discussion about `ros2_control_node` and its time management is again actual in #640.
I am still not sure if we are measuring the correct period we would like to have.
From my understanding, we should measure time between `update` calls without inclusion of `read` and `write` methods.
I suggest a bit updated time measurement with the following snippet:

```
  // initialize variables used in while
  rclcpp::Time begin = cm->now();
  rclcpp::Time end = begin;
  rclcpp::Time current_time = begin;
  rclcpp::Time previous_time = current_time;

  // Use nanoseconds to avoid chrono's rounding
  std::this_thread::sleep_for(std::chrono::nanoseconds(1000000000 / cm->get_update_rate()));
    
  begin = cm->now();
  while (rclcpp::ok())
  {
    cm->read();
    current_time = cm->now();
    cm->update(current_time, current_time - previous_time);
    cm->write();
    previous_time = current_time;
    end = cm->now();
    std::this_thread::sleep_for(std::max(
      std::chrono::nanoseconds(0),
      std::chrono::nanoseconds(1000000000 / cm->get_update_rate()) -
        std::chrono::nanoseconds(end.nanoseconds() - begin.nanoseconds())));
    begin = cm->now();
  }
});
```

_Originally posted by @destogl in https://github.com/ros-controls/ros2_control/issues/568#issuecomment-955158952_

AndyZe, 2022-02-12:
I'm not sure I understand. Is your thinking basically along these lines?

Internal controller time is tracked within update(). Therefore the sleep in the main control loop should only depend on the update function.

AndyZe, 2022-02-12:
Side note: the code you posted isn't exactly right because `begin` isn't updated anywhere. I think I get the idea, though.

closed 2022-02-16 by bmagyar