ros-controls/ros2_control #649: Hardware interface: variable update rates for command/state interfaces

bijoua29, 2022-02-16:
I wanted to bring up an ros2_control issue for us that may or may not be unique to us. The details below might also expose some of my ignorance of ros2_control so feel free to point that out. I also apologize for the lengthy text.

**Is your feature request related to a problem? Please describe.**
We have a hardware device that is bandwidth constrained. As such, we have to be judicious about how often we interact with the hardware. The ros2_control controller manager runs at a fixed rate, lets say 200 Hz. That means it calls into the hardware interface read() and write() methods at 200 Hz. I understand that we have the ability to set individual update rates(at a controller manager update rate divisor) for controllers but that just means the controller writes and reads from the interfaces at this rate. The controller manager still reads/writes to the hardware interface at 200 Hz - at least, I think that is the case. In any case, semantically it may not make sense to break up a controller into separate controllers just to run them at different update rates.

The problem that we run into is that not all interfaces, state or command, need to run at full 200 Hz rate. There may be environmental state e.g. temperature, current, hw status, etc. that may only be needed at 1 Hz. Similarly, a command interface that sends a watchdog to the hardware may only need to run at 1 or 2 Hz, for example. The only command interface that needs to run at full rate of 200 Hz is the actuation interface e.g. a position or velocity command. Now, we could run all interfaces at 200 Hz with the downside of efficiency and wasted cycles. However, in some cases, as in the hardware device I alluded to earlier, running all interfaces at 200 Hz would oversubscribe the available bandwidth, resulting in retries and dropped packets and added latency. So, in this case, we have to tune the update rate for each interface, state or command, such that the total traffic to/from the device is less than the available bandwidth.

**Describe alternatives you've considered**
One way, to resolve this issue, is to meter the traffic to the device in the hardware interface. We have done this for other drivers. What this involves is only sending some command/state requests to the hardware at some divisor of the controller manager rate. One issue is that the hardware interface is unaware of the controller manager update rate so if the hardware interface wants to request the temperature at, say, 1 Hz, then the controller manager update rate will need to be hardcoded into the hardware interface and the temperature is only requested 1 out every 200 calls into the read() method. I don't find this approach very clean since I have to manage several variables that keep track of the different update rates needed by the hardware interface as well as hardcode the controller manager update rate. Also, normal constructs such as timers and threads are typically not available to a hardware interface, although I presume you could hack something together. I would rather have some higher level configuration keep track of the update rates needed by each command/state interface and the hardware interface just deal with processing command and state interface requests.

**Describe the solution you'd like**
There can be two ways that the controller manager can help with this issue. One approach would be to specify additional divisor rates for the controller manager so that it calls into the hardware at different rates perhaps with an argument indicating the rate or period. In this scenario, the hardware interface would still have to manage which interfaces need to be called at which rate but it doesn't have to do the divisor logic in the driver and maintain variables to manage this. Another approach is to optionally set the update rate for each interface in configuration. The controller manager would call into the hardware interface at each update rate with the set of interfaces for that update rate. In this scenario, the hardware interface doesn't have to deal with the update rates and just processes the interfaces specified in the read()/write() calls. The second approach is much preferable.

**Additional context**
I'm aware that this potentially brings up issues about the interaction between the controller and the command/state interfaces which might be at a different update rate. There also might be limitations within the current ros2_control architecture that preclude a clean solution to this problem. There also might be other mitigations within ros2_control that I am not aware of and might help with this problem - if so, please let me know what potential solutions might already exist or solutions that others have used.

coxep, 2024-07-23:
Strongly second this; I have some interfaces that need to be updated very infrequently and others that need to run at a much higher rate. It seems inefficient to slow down the high-rate loop by being forced to read all states for every read call. Would much rather receive a vector of strings capturing the specific states a specific controller is requesting.

saikishor, 2024-07-23:
> Strongly second this; I have some interfaces that need to be updated very infrequently and others that need to run at a much higher rate. It seems inefficient to slow down the high-rate loop by [...]

This should now be possible with the combination of async and different read/write rates at hardware level. There are open PRs regarding the same. One of this is: https://github.com/ros-controls/ros2_control/pull/1570. Do you think, it solves your usecase?

coxep, 2024-07-23:
> This should now be possible with the combination of async and different read/write rates at hardware level. There are open PRs regarding the same. One of this is: #1570. Do you think, it solves your [...]

Not quite. I have a single hardware interface that provides a variety of states, utilized by a variety of controllers. Some controllers need some of the states from this single hardware interface at 100Hz, and others need some of the states from that same hardware interface at 0.1 Hz. My issue is that I don't like the hardware interface having to read ALL the states at 100Hz.

I can of course implement logic in the hardware interface "read" call to ensure that some states are read from hardware at appropriate rates, but think it would be more elegant for each controller to request information from the hardware interface directly (with a mapping to the states that specific controller needs).

saikishor, 2024-07-23:
I don't understand the problem here. If you don't update the State and Command interfaces at same rate, it will reuse the same value instead. So, you are really not obliged at all.

For instance, a controller is reading a State Interface at 100Hz but the Hardware is updating it at 10Hz. This means for 10 controller update cycles the state will remain with the same value.

coxep, 2024-07-23:
> I don't understand the problem here. If you don't update the State and Command interfaces at same rate, it will reuse the same value instead. So, you are really not obliged at all.
>
> For instance, a [...]

I'll break it down a bit better. Single hardware interface, with a single "read" call for all states managed by that interface.

Some controllers (differential drive controller) need position states from the hardware_interface at a high rate, and other controllers (battery) need voltage data from the same hardware_interface at a very low rate. But the same call (read) is used by hardware_interface to provide all available states, and this runs at a fixed rate (100hz). I am happy providing position at 100Hz. I'm not happy with inefficiently reading voltage at 100Hz.

christophfroehlich, 2024-07-23:
https://discourse.ros.org/t/best-practices-for-accessing-non-controls-information-alongside-ros2-control/38721/6?u=christophfroehlich

saikishor, 2024-07-23:
Then it's implementation detail. It's upto you to separate them into different components or if this is not possible. You can skip the reading of voltage by internal filtering like updating the voltage value every 100 cycles or whatever is required by a simple counter. This way you will avoid updating voltage with same frequency and maintain the same value throughout these skipped cycles

coxep, 2024-07-23:
> Then it's implementation detail. It's upto you to separate them into different components or if this is not possible. You can skip the reading of voltage by internal filtering like updating the [...]

I don't dispute this. To reiterate my earlier comment:

> I can of course implement logic in the hardware interface "read" call to ensure that some states are read from hardware at appropriate rates, but think it would be more elegant for each controller [...]

This is also nicely covered in documentation: https://control.ros.org/rolling/doc/ros2_control/hardware_interface/doc/different_update_rates_userdoc.html

My original post is simply a vote for @bijoua29 's enhancement request :)

Rather than putting extra logic into the hardware_interface, I think it would be nice/elegant from a user perspective to offload rate control to the controllers themselves (again, a vector of states might be passed to the `read` function in the hardware_interface, and the read function is then expected to update this subset of states)

closed 2025-07-07 by github-actions[bot]