ros-controls/ros2_control #654: Controller manager compilation fails on galactic

Minipada, 2022-02-21:
**Describe the bug**
Controller manager compilation fails.

```
--- stderr: controller_manager
CMake Error at /opt/ros/galactic/share/ament_cmake_python/cmake/ament_python_install_package.cmake:39 (message):
  ament_python_install_package() called with unused arguments:
  SCRIPTS_DESTINATION;lib/controller_manager
Call Stack (most recent call first):
  /opt/ros/galactic/share/ament_cmake_python/cmake/ament_python_install_package.cmake:33 (_ament_cmake_python_install_package)
  CMakeLists.txt:156 (ament_python_install_package)

---
Failed   <<< controller_manager [4.11s, exited with code 1]
```

**To Reproduce**
Steps to reproduce the behavior:
1. Compile latest source code of controller_manager 223b2593b09e3c2b453d8c02f7f402d590316124

**Expected behavior**
Compiles

**Screenshots**
[image]

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Version Galactic

**Additional context**
Seems this line is broken https://github.com/ros-controls/ros2_control/blob/223b2593b09e3c2b453d8c02f7f402d590316124/controller_manager/CMakeLists.txt#L156
linked to this commit https://github.com/ros-controls/ros2_control/commit/a3f179ce337a780575892e9c87cf93315e669ef4

destogl, 2022-02-24:
This is not a bug, but intended behavior. You are using the wrong branch... ROS" Galactic == 'galactic' branch

And with this answer, I earned myself a beer. (I see you are located in Hamburg, if you send it today via DHL it should be in Karlsruhe tomorrow just in time for after-work beer) :smile:

closed 2022-02-24 by destogl

Minipada, 2022-02-24:
Hi, I saw that after, forgot to mention it here 🙃, my bad. Unfortunately (or not?) not I'm in Brazil, not in Hamburg until end of march. If you pass by, or I come to Karlsruhe, I'd be down for a beer 😉
[image]

rcywongaa, 2022-03-03:
Should the Dockerfile reflect this?  Currently [`[galactic] Dockerfile`](https://github.com/ros-controls/ros2_control/blob/galactic/.docker/source/Dockerfile) still uses the master branch `ros2_control.repos` which doesn't seem right.

Somewhat relatedly, shouldn't [`[master] Dockerfile`](https://github.com/ros-controls/ros2_control/blob/master/.docker/source/Dockerfile) set `ROS_DISTRO="rolling"`  since if I understand correctly, master branch targets rolling?

I can open an issue and submit a PR if my understanding is correct