ros-controls/ros2_control #665: Add controller type to load controller service

jlack1987, 2022-03-07:
Is this something that has been considered before? I feel like it's a very small price to pay for something that would really simplify the overall process of loading a controller especially with the way parameters are different in ros2 as opposed to ros1.

destogl, 2022-03-07:
I don't see anything against it. Can you explain a bit more your use-case?

destogl, 2022-03-10:
Yes, we could add this. Do you mind open a PR with this functionality?

We should not remove anything because there are many cases where you want to simply use controller names and type is “not essential”. So, service should have type as an optional parameter. If type-argument is provided, [other `load_controller` method](https://github.com/ros-controls/ros2_control/blob/6b495ef86889cbfdc1ee22ac77efa2ff589727ee/controller_manager/src/controller_manager.cpp#L229) can be called that accepts booth.