ros-controls/ros2_control #668: Variables being declared inside realtime functions / loops

AndyZe, 2022-03-08:
I thought it was bad practice to declare a variable inside a realtime loop, but I see it a lot in this codebase. For example, [here](https://github.com/ros-controls/ros2_control/blob/6b495ef86889cbfdc1ee22ac77efa2ff589727ee/hardware_interface/src/actuator.cpp#L235) in actuator.cpp `write()`:

```
return_type Actuator::write()
{
  return_type result = return_type::ERROR;
```

Compare to the ROS1 ros_control codebase where a lot of effort went into avoiding this practice. For example, [here's](https://github.com/ros-controls/ros_control/blob/83a392bafa1ec01dc77ec2a592f3b74a783f3faa/hardware_interface/include/hardware_interface/actuator_command_interface.h#L58) the equivalent of `write()`:

`void setCommand(double command) {assert(cmd_); *cmd_ = command;}`

Are we really trusting compilers so much these days or am I missing something?

gavanderhoorn, 2022-03-09:
Anything which goes on the stack isn't really a problem, as it can be allocated deterministically (it's essentially `O(1)`, as it's simple pointer arithmetic).

Memory allocation from a heap is something which is not inherently deterministic (at least not with the default memory allocators), and that's what you should avoid when writing code which should be deterministic.

If any of the classes pointed out have member variables which (might) allocate on the heap (such as `std::string` and any of the container classes), that might be a problem and would have to be audited.

Alternatively, a deterministic memory allocator could be used (such as TLSF).

destogl, 2022-03-10:
We should agree on the best approach/rules in this issue and open a bunch of good-first-issues to clean the code base completely. It is true that some places need a lot of house-keeping.

AndyZe, 2022-04-26:
I'm planning to start working on this issue of resizing variables in the joint trajectory controller `update()`. The only two strategies I'm aware of are:

- Pre-allocate a large-ish amount of memory ahead of time, then only use what we need
- Use a deterministic memory allocator (TLSF) as mentioned by gvanderhoorn

Any preferences or other ideas?

closed 2025-07-07 by github-actions[bot]