ros-controls/ros2_control #671: Ros2_control diff_drive cant turn in gazebo

RobertForristall, 2022-03-10:
**Describe the bug**
When working with the ros2_control package using the diff_drive_controller in gazebo the robot responds to the linear.x component of the twist message and drives forward; however it doesn't respond to the angular.z component of the message for turning.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the following package: https://drive.google.com/drive/folders/1SJmruuz5IH7Jz0EOMfMslK-AsFCxvWfa?usp=sharing

Important files in package:
- urdf of robot: resource/ezrassor.xacro
- gazebo file for robot: resource/ezrassor.gazebo
- ros2_controller config file: config/standard_rover/default_position_controllers.yaml
- launch file for simulation/robot: launch/simulation_launch.py

2. Build package:
- colcon build

3. Run the following launch command: 
- ros2 launch ezrassor_arm_v2 simulation_launch.py rover_model:=standard control_methods:=autonomous

4. Wait for all nodes to launch
- This will finish with a log to the console saying that the object detection and autonomous control nodes initalized.

5. Publish the following message to console which will cause the robot to drive forward as intended:
- ros2 topic pub /ezrassor/diff_drive_controller/cmd_vel_unstamped geometry_msgs/Twist  '{linear:  {x: 1.0, y: 0.0, z: 0.0}, angular: {x: 0.0,y: 0.0,z: 0.0}}'

6. Publish the following message to console which will show the error as the robot will not turn as intended:
- ros2 topic pub /ezrassor/diff_drive_controller/cmd_vel_unstamped geometry_msgs/Twist  '{linear:  {x: 0.0, y: 0.0, z: 0.0}, angular: {x: 0.0,y: 0.0,z: 1.0}}'

**Expected behavior**
The robot should turn or at least show force being exerted to the wheels in an attempt to turn but no response is seen from the tires at all when given the angular.z component

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04 (Focal Fossa)
 - Version Foxy

**Additional context**
Seems that there may be an issue with the communication of the angular.z component to the wheel drivers in gazebo. It could be an issue with the controller it self or gazebo but i figured this would be the best place to start since someone else may have experience with this problem

destogl, 2022-03-10:
Hi, did you check this answer: https://answers.ros.org/question/395015/differential-drive-not-turning-the-vehicle/#396956?

Since you are using foxy, I don't guarantee anything about this working properly. I didn't test it there. Maybe some of the essential recent changes are missing.

P.S. ros2_controllers would be better place to open this issue.

destogl, 2022-03-12:
> I have gone ahead and opened an issue for it on ros2_controllers to see if they can give any insight.

There is no “they” those are all the same people. For the rest of the discussion, check there: ros-controls/ros2_controllers#312

closed 2022-03-12 by destogl