ros-controls/ros2_control #676: base_interface file missing in galactic branch  ros2_control/hardware_interface/include/hardware_interface/

HusamZain, 2022-03-12:
Hei!
thank you for your work.
I'm using ros2 galactic. and trying to use[ this ](https://github.com/Factor-Robotics/odrive_ros2_control.git)
odrive package with ros2-control framework, however when I try to build the package I got this error 
[image]
when I checked your galactic branch I found out that there is no file called base_interface.hpp
so I was wondering if there is a way to solve this issue ?

HusamZain, 2022-03-14:
I changed all the return_type that was proposed in the documentation (configure, active ,deactive), but wasn't sure about the 'prepare_command_mode_switch', and if I do change it I got this error, sorry for the inconvenient 
[image]
any suggestions ?

destogl, 2022-03-14:
Hi,

you should not change the return type of the `prepare_command_mode_switch`. If you are unsure, always check the header files of parent classes. Usually, an IDE can show you this (if configured properly).

closed 2022-03-14 by destogl