ros-controls/ros2_control #688: [ros2_control CLI] `set_controller_state` should support full lifecycle and use lifecycle nomenclature.

destogl, 2022-03-23:
## Background

set_controller_state CLI command is not full. There are missing some states from there, and we should use lifecycle nomenclature there instead of old one, e.g., `activate` and `deactivate` instead of `start` and `stop`.

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/galactic/Tutorials/Workspace/Creating-A-Workspace.html), for Step3 use "Download Source Code" section with [these instructions](https://ros-controls.github.io/control.ros.org/getting_started.html#compiling).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://ros-controls.github.io/control.ros.org/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/galactic/Tutorials.html)
* [ROS Answers](https://answers.ros.org/questions/)

**Good luck with your first issue!**

adhithiyan110, 2025-05-10:
Hi, I'd like to work on this issue as my first open source contribution

KroNton, 2025-08-28:
Hi @destogl @christophfroehlich , 
I'd like to work on this issue. Can I take it?

christophfroehlich, 2025-08-28:
@KroNton sure! Please evaluate first if this is still relevant, as the issue is more than 3 years old. Before working on it, write a short proposal what you will change and we can discuss this first.

KroNton, 2025-08-28:
Hi @christophfroehlich ,

Thank you for the feedback!

I’ve reviewed the current implementation of `set_controller_state.py` , and while it supports basic functionality, I believe issue #688 is still relevant because:

- The CLI uses target states (inactive, active) instead of transition commands (activate, deactivate, etc.)
- It’s inconsistent with ros2 lifecycle set, which uses verbs.
- Transitions like configure and cleanup are not directly accessible.
- There’s a long-standing TODO about implementing cleanup in the CLI.
### I propose to:
1. Replace `choices=['inactive', 'active'] ` with `choices=['configure', 'activate', 'deactivate', 'cleanup']`
2. Uncomment and use cleanup_controller() for cleanup
### Example:
```bash
ros2 control set_controller_state my_controller activate
ros2 control set_controller_state my_controller cleanup
```
This would make the CLI more intuitive, consistent, and complete.

Please let me know if this direction works for you — happy to adjust!

Thanks!

christophfroehlich, 2025-12-25:
Hi @KroNton, sorry that you've got no response to your suggestions. 
Fyi: The cleanup_controller service just got added to the framework, it didn't exist before.

Regarding the initial post in this issue: I think the verb is now "complete" in the sense that every target state can be reached (by also using load/unload_controller), and its choices are conformant with the ros2 lifecycle design target state names.
The only question is: Should we change the CLI to match the ros2lifecyclecli and use transition names as arguments instead? I am not 100% sure about that, I would rather add a new verb like `trigger_transition` and use the transitions as choices.

Opinions about that? @soham2560 or @saikishor maybe?

saikishor, 2025-12-26:
> The only question is: Should we change the CLI to match the ros2lifecyclecli and use transition names as arguments instead? I am not 100% sure about that, I would rather add a new verb like [...]

If there is no clear necessity, i would avoid adding new code.

We can definitely improve the `set_controller_state` verb to add `cleanup_controller` service, when called with `inactive` state and the controller is also in `inactive` state. Does it make sense? What do you think?

christophfroehlich, 2025-12-31:
I support the idea to have similar syntax to ros2lifecyclecli, but as mentioned, I am not 100% happy with the verb name then. but we could add them to the current verb, just calling the transition without checking the current state.

christophfroehlich, 2025-12-31:
yes, similar to ros2lifecyclecli

saikishor, 2025-12-31:
> yes, similar to ros2lifecyclecli

Got it