ros-controls/ros2_control #691: [Interfaces] Make functionality of `get_name()` method clear.

destogl, 2022-03-24:
## Background

In `hardware_interface/handle.hpp` file we have three ways to get name of the interface, joint and full name. The naming of methods is a bit confusing:

- `get_name()` - returns "joint" name
- `get_interface_name()` - return interface name (OK)
- `get_full_name()` - returns full name of the interfaces `<joint_name>/<interface_name>`

I find this a bit confusing and propose the following change:

- `get_name` - returns full name of the interfaces `<joint_name>/<interface_name>`
- `get_prefix_name` - returns "joint" name (better is to call this prefix instead of joint)

Tasks:
- [ ] update functionality in `handle.hpp` file
- [ ] update functionalities in `loaned_*_interface.hpp` files
- [ ] update functionalities in tests (and write tests to check all three methods if those are missing)
- [ ] update ros2_controllers repository if any of the methods is used.

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/galactic/Tutorials/Workspace/Creating-A-Workspace.html), for Step3 use "Download Source Code" section with [these instructions](https://ros-controls.github.io/control.ros.org/getting_started.html#compiling).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://ros-controls.github.io/control.ros.org/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/galactic/Tutorials.html)
* [ROS Answers](https://answers.ros.org/questions/)

**Good luck with your first issue!**

Schulze18, 2022-04-14:
Hello everyone, I would like to work on this one. 

@destogl, just to make it clear, it is not necessary any changes in get_interface, right? Therefore, as a final result, we would have:
- `get_name()` : returns full name of the interfaces `<joint_name>/<interface_name>`
- `get_interface_name()` : returns interface name `<interface_name>`
- `get_prefix_name()` : returns "joint" name (better is to call this prefix instead of joint) `<joint_name>`

destogl, 2022-04-20:
> Hello everyone, I would like to work on this one.

Yes, exactly, this should be the final solution. For the first implementation, we should probably deprecate `get_full_name()` method using `get_name` in the background.

@Schulze18 I will assign you for this than! Thanks! Please ask if you have any further questions!

Schulze18, 2022-04-28:
I have a few doubts. @destogl, to deprecate `get_full_name()`, I just made it return `get_name()` and replaced any occurrence to `get_name()`. Should I add any other comment or message to indicate it was deprecated? 

Sorry, it is my first time going deep into the ros2 code. Do the classes [ResourceStorage](https://github.com/ros-controls/ros2_control/blob/5363e3a248e5fc6379e44cd50d6fe59cc05514a3/hardware_interface/src/resource_manager.cpp#L66) and [ResourceManager](https://github.com/ros-controls/ros2_control/blob/5363e3a248e5fc6379e44cd50d6fe59cc05514a3/hardware_interface/src/resource_manager.cpp#L520) inherit ReadOnlyHandle? There are many `hardware.get_name()` and `component.get_name()` that I am not sure if I need to change to `.get_prefix_name()`.

bmagyar, 2023-12-21:
Yup!

closed 2023-12-21 by bmagyar