ros-controls/ros2_control #692: Checking if interfaces are claimed with the ControllerManager

shonigmann, 2022-03-30:
Is it possible to directly check whether a command interface is claimed through the ControllerManager?

I'd like to work on a fix to ros-simulation/gazebo_ros2_control#85, where the Gazebo ROS2 Control Plugin is unable to support more than one command interface type even when only one is claimed at a time.  

The most direct approach feels like it would involve querying the ResourceManager's [`command_interface_is_claimed()`](https://github.com/ros-controls/ros2_control/blob/a11638d9e3c4acf675ad633bd9854efe8351e7cc/hardware_interface/src/resource_manager.cpp#L612-L621) method, but the resource_manager_ is private within the ControllerManager. 

I was thinking of putting the check in the Gazebo plugin's [`Update()`](https://github.com/ros-simulation/gazebo_ros2_control/blob/698d01381423cea3232a78ab0ada7eb12cd6d40b/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp#L364-L380) loop, so I was hoping to avoid using the service interface which feels like it would be too slow. I'm not aware of any events that I could subscribe to that would let me monitor for changes in claimed resources, but would be open to taking that route if its available. 

And hopefully this isn't out of place on the issue tracker. I can move this over to ROS Answers if that would be more appropriate.

shonigmann, 2022-04-01:
For the time being I am using the ControllerManager's ListControllers service interface, which does provide the desired functionality... That said it takes ~1ms to call, so its definitely not ideal to run inline with the update loop. An event based approach would be wonderful if it exists.

destogl, 2022-04-03:
Hi,

This is a correct place for this, because it is a bug fix on the framework. 

This should be managed by internal flags that are set in "prepare_command_mode_switch" and "perform_command_mode_switch". There you have exact information which interfaces are used in controller that will be activated or deactivated. This being said, you know exactly hoch interfaces are exactly used in any given moment of time.

Any other approach is mixing layers of the framework I
And it is not guaranteed to work as intended.

destogl, 2022-04-06:
> Just to clarify, are you suggesting that the aforementioned flags need be added to the base [`Actuator`](https://github.com/ros-controls/ros2_control/blob/a11638d9e3c4acf675ad633bd9854efe8351e7cc/hardware_interface/src/actuator.cpp) / [`System`](https://github.com/ros-controls/ros2_control/blob/d58ecf1dc981aa80b63eb2aa7195730237a3da65/hardware_interface/src/system.cpp) implementations?

I was also thinking about hat, but since flags are dependent on the hardware and its implementation, I am uncertain if we can make a sensible generic solution to it.

Let us know if you have some ideas about generalizing your implementation. You will for sure have more knowledge/experience about it.

closed 2025-07-07 by github-actions[bot]