ros-controls/ros2_control #698: Use topic for the robot description parameter

joe28965, 2022-04-12:
**Is your feature request related to a problem? Please describe.**
Right now I'm making a physical robot that also has a 'digital twin'.

For gazebo_ros2_control it's necessary to add the gazebo hardware interface to the URDF, while ros2_control needs the real one (makes sense). Right now I've set up a xacro argument to choose between gazebo and real hardware interface.

The thing is, I can do this and then use robot_state_publisher to publish the correct URDF. The gazebo_ros spawner and gazebo_ros2_control both just subscribe to the topic to get the copy of my URDF. Ros2_control is the only one that requires its own copy of the URDF file as a parameter. 
Is there a specific reason for this? Would it not be easier to receive the URDF from robot_state_publisher or am I missing some obvious reason as to why the current way is better?

**Describe the solution you'd like**
To have the option to use a topic to supply the URDF.

**Describe alternatives you've considered**
Right now I call xacro again in my launch file for ros2_control and add the result to my controller manager. It's not a deal breaker by any stretch of the imagination, it would just be nice to have the option for a topic.

joe28965, 2022-04-20:
With regards to the URDF being updated.
As I mentioned gazebo_ros2_control uses robot_state_publisher to get the URDF, so I decided to check how they get the URDF. That actually made me realize I'm wrong (partially).

During the init gazebo_ros2_control actually requests the _parameter_ from robot_state_publisher ([the function in question, if you're interested](https://github.com/ros-simulation/gazebo_ros2_control/blob/698d01381423cea3232a78ab0ada7eb12cd6d40b/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp#L390)). It doesn't actually subscribe to the topic. As far as I can see, it does this once, during the init.

That could be a solution for CM too? Request the parameter from robot_state_publisher once, instead of requesting it from it's own list of parameters. That would mean it is not possible to dynamically reconfigure the URDF. However, I don't know if that's possible in the CM now or if that is desired.

As for the final comment. I have CM in a separate launch file from gazebo, robot_state_publisher and gazebo_ros spawner. Would substitutions like that work over separate launch files?

destogl, 2022-04-20:
> As for the final comment. I have CM in a separate launch file from gazebo, robot_state_publisher and gazebo_ros spawner. Would substitutions like that work over separate launch files?

Not they wouldn't. sorry didn't know you exact setup.

> That could be a solution for CM too? Request the parameter from robot_state_publisher once, instead of requesting it from it's own list of parameters. That would mean it is not possible to [...]

Thanks for your input and information! We could do proposed approach, but I would prefer if we think long-term and do this a bit more intelligent also for the future use. TBH I see uses where we would update robot description during run-time, e.g., dynamic adding of hardware and similar. What do you think about following proposal: (also @bmagyar and @livanov93 what do you think)

- We can update robot description on runtime.
- If there is a new hardware we don't know in Controller Manager (Resource Manager more specifically) we automatically load its plugin (maybe even start it - do we need an additional parameter for this?)
- if hardware that is in Controller Manager is removed, then we should check its state. If it is in INACTIVE or ACTIVE state we give a warning about that and hardware has to be unloaded manually.
- if hardware is in UNCONFIGURED state then we unload it automatically

This could be not-so-bad first functionality.

joe28965, 2022-04-21:
> e.g., dynamic adding of hardware and similar.

In that case, there is a chance robot_state_publisher would need to be updated too, so using the same parameter would be easier for the user in this case. However:

> We can update robot description on runtime.

If using the parameter from RSP, we'd need to know when that one is being updated. Since periodically requesting it seems like a bloated solution. Is there a way to get informed when the parameter of _another node_ is updated?

> (maybe even start it - do we need an additional parameter for this?)

Might make sense to start it, since plugins are now also automatically started when loaded (at least by default).

> if hardware that is in Controller Manager is removed, then we should check its state. If it is in INACTIVE or ACTIVE state we give a warning about that and hardware has to be unloaded manually.

Or would it be possible to take action ourselves here, especially if it's INACTIVE? In the way of "the hardware is removed, so we'll just unload it". Or does that fall into the category "doing too much without developer input"?

bmagyar, 2022-04-22:
I think this issue developed beyond the original scope, probably time to close it and start a discussion on a new thread re. dynamic handling of updating robot_description.

Is there anything that could be salvaged from the original discussion?

joe28965, 2022-04-22:
The original issue started with the use of the robot_state_publisher parameter for the URDF instead of a separate parameter. Think that's the main thing?

destogl, 2022-04-24:
> If using the parameter from RSP, we'd need to know when that one is being updated. Since periodically requesting it seems like a bloated solution. Is there a way to get informed when the parameter of another node is updated?

What I am thinking that we switch to using topics completely for robot description. This means that `robot_state_publisher` decides on changes when this has to be changed (hopefully across all nodes, then. I know that MoveIt2 had also discussions about this.)

> Or would it be possible to take action ourselves here, especially if it's INACTIVE? In the way of "the hardware is removed, so we'll just unload it". Or does that fall into the category "doing too much without developer input"?

This would be possible, we just have to decide what makes sense.

@bmagyar I would rather just rename the issue. Or @joe28965 if you like, you can open the new issue where you could summarize all details from this discussion, so we can start to work on the implementation.

joe28965, 2022-04-25:
> This would be possible, we just have to decide what makes sense.

Personally it doesn't make sense to me to update RSP with a new URDF that removed hardware, without wanting to remove it from ros2_control. Why was the hardware removed from the update to RSP if it's still being actively used. I suggest unloading regardless of the state.

**Summary:** (think this covers all, if you agree I'll make a new issue for it)

# Use the robot_state_publisher topic to receive the URDF, allow for updating on runtime

By using the robot_state_publisher to receive the URDF, there will be a more centralized place to update the URDF that multiple nodes can use (instead of multiple nodes all keeping track of their own URDF).

- Update the robot description if the URDF from robot_state_publisher is updated.
- If there is new hardware that the Controller Manager (or Resource Manager more specifically) doesn't know, we automatically load the plugin (and start it too? Do we need an additional parameter to choose the whether or not to start?).
- If hardware that is in the CM is removed from the updated URDF, then we unload it automatically.

joe28965, 2022-04-28:
I've created a new issue at #711 with the summarized text. I'll close this issue to keep any future discussion localized to 1 issue.

closed 2022-04-28 by joe28965