ros-controls/ros2_control #725: hardware_interface in Rolling fails to build on RHEL 8 target

nuclearsandwich, 2022-05-26:
**Describe the bug**

Build failures in ROS 2 Rolling build farm on RHEL/AlmaLinux 8

**To Reproduce**

https://build.ros2.org/view/Rbin_rhel_el864/job/Rbin_rhel_el864__hardware_interface__rhel_8_x86_64__binary

**Environment (please complete the following information):**
 - OS: RHEL / AlmaLinux 8
 - Version: Rolling, ros2_control v2.9.0-1

**Additional context**

My C++ appetite isn't enough to tell at a glance if this is an uncaught dependency or a C++/compiler version or flag issue.

We don't currently have devel jobs for RHEL 8 configured but if you need help setting up a Docker image for testing we can give some pointers. Another option is to disable the build for RHEL if fixing it is not a high priority.

bmagyar, 2022-05-30:
Happy to leave the issue open? We have RHEL in our CI but rarely follow up on it. We do intend to however...

bmagyar, 2022-06-28:
I'm closing this one as #730 fixed what we could in the CI. @nuclearsandwich we are happy to address further things but hopefully the CI is representative

closed 2022-06-28 by bmagyar