ros-controls/ros2_control #732: joint_state_broadcaster configuration fails

christophfroehlich, 2022-06-09:
This is a repost of https://github.com/ros-controls/gazebo_ros2_control/issues/131 because it seems to be linked with the [changed hardware interface](https://github.com/ros-controls/gazebo_ros2_control/issues/131#issuecomment-1145258747) logic, hoping that it gets some attention here.

**Describe the bug**
I tried to use the humble or rolling releases and struggle with launching some robots. 
While `ros2 launch ros2_control_demo_bringup rrbot.launch.py` from [ros2_control_demos](https://github.com/ros-controls/ros2_control_demos.git) works, some others using gazebo don't, e.g. `ros2 launch ros2_control_demo_bringup rrbot_system_position_only_gazebo.launch.py` throws
> [gzserver-1] [INFO] [1653938781.009092923] [controller_manager]: Configuring controller 'joint_state_broadcaster'
> [gzserver-1] [INFO] [1653938781.010581373] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
> [gzserver-1] [ERROR] [1653938781.036334316] [joint_state_broadcaster]: None of requested interfaces exist. Controller will not run.
> [gzserver-1] [WARN] [1653938781.036404912] []: Error occurred while doing error handling.
> [gzserver-1] [ERROR] [1653938781.036445872] [controller_manager]: After activating, controller 'joint_state_broadcaster' is in state 'unconfigured', expected Active

Same error with the example `ros2 launch gazebo_ros2_control_demos cart_example_velocity.launch.py` from [gazebo_ros2_control_demos](https://github.com/ros-simulation/gazebo_ros2_control).

I can't find any differences in the `joint_state_broadcaster` configuration (yaml, launch-files, urdf). 
Is this linked to ros2_controllers, ros2_control, gazebo_ros2_control, or the example files?

**To Reproduce**
Steps to reproduce the behavior:
1. Install ros2-rolling on Ubuntu 22.04
2. Download [ros2_control_demos](https://github.com/ros-controls/ros2_control_demos.git) and [gazebo_ros2_control](https://github.com/ros-simulation/gazebo_ros2_control.git)
3. Install its dependencies with rosdep and compile it
4. Run `ros2 launch ros2_control_demo_bringup rrbot_system_position_only_gazebo.launch.py`
5. See error

**Environment:**
I manually compiled humble on Ubuntu 20.04 with the depencies found in the [release repositories for humble](https://github.com/ros2-gbp/ros2_control-release).
Additionally, I tried a freshly installed Ubuntu 22.04 with ros2-rolling (and humble, but there are some releases missing or rather old releases with the debians) with the same errors.

destogl, 2022-06-28:
Hi @christophfroehlich, sorry I was on vacation and didn't see this issue earlier. 

The issue you mention is on `gazebo_ros2_control` side and not joint state broadcaster. I will skip to the discussion there and close this here.

closed 2022-06-28 by destogl

tonynajjar, 2023-11-09:
Hello, this has now resurfaced; the symptoms are the same but the root cause is probably different. I'm getting

```
taskset-9] [INFO] [1699541862.566469535] [controller_manager]: update rate is 50 Hz
[taskset-9] [INFO] [1699541862.569287305] [controller_manager]: RT kernel is recommended for better performance
[taskset-9] [INFO] [1699541864.169134359] [controller_manager]: Loading controller 'joint_state_broadcaster'
[taskset-9] [INFO] [1699541864.787324074] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[taskset-9] [INFO] [1699541864.787608668] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[taskset-9] [ERROR] [1699541864.869637610] [joint_state_broadcaster]: None of requested interfaces exist. Controller will not run.
[taskset-9] [WARN] [1699541864.869827734] [joint_state_broadcaster]: Error occurred while doing error handling.
[taskset-9] [ERROR] [1699541864.869903354] [controller_manager]: After activation, controller 'joint_state_broadcaster' is in state 'unconfigured' (1), expected 'active' (3).
[taskset-9] [INFO] [1699541864.944790808] [controller_manager]: Received robot description from topic.
[taskset-9] [INFO] [1699541864.946137930] [resource_manager]: Loading hardware 'AmrHardware' 
[taskset-9] [INFO] [1699541864.950150557] [resource_manager]: Initialize hardware 'AmrHardware' 
[taskset-9] [INFO] [1699541865.086481513] [resource_manager]: Successful initialization of hardware 'AmrHardware'
[taskset-9] [INFO] [1699541865.086653843] [resource_manager]: 'configure' hardware 'AmrHardware' 
[taskset-9] [INFO] [1699541865.086721463] [resource_manager]: Successful 'configure' of hardware 'AmrHardware'
[taskset-9] [INFO] [1699541865.086744473] [resource_manager]: 'activate' hardware 'AmrHardware' 
[taskset-9] [INFO] [1699541865.086771738] [AmrHardware]: Successfully activated!
```

I'm on `3.20.0`, do you see any recent change that could have caused this?

saikishor, 2023-11-09:
Hello!

Are you launching this joint_state_broadcaster using the command line or using the launch file? If this is the case, it is expected that it might happen, as when loading the controller the `generate_parameter_library` declares the parameters and then retrieves them. However, the parameters won't exist until we load the controller and then load the parameters using `--param-file` argument. 

In order to avoid this, you can parse the controller parameter upon launching the ros2_control_node as below, this does solve the issue for now. We are trying to fix this behavior in the future.

````
control_node = Node(
        package="controller_manager",
        executable="ros2_control_node",
        parameters=[robot_description, <robot_controllers_param_file>],
        output="both",
    )
    ````

tonynajjar, 2023-11-09:
I'm launching it using the launch file, I also already pass the param file as a parameter to the ros2_control_node :thinking: (not the robot_description as it gets it from the topic)

saikishor, 2023-11-09:
> I'm launching it using the launch file, I also already pass the param file as a parameter to the ros2_control_node 🤔 (not the robot_description as it gets it from the topic)

So you passed the param file that has the `joint_state_broadcaster` parameters as well as its ros parameters?

tonynajjar, 2023-11-09:
exactly, here is what it looks like:

amr_controllers.yaml
```
/**:
  controller_manager:
    ros__parameters:
      update_rate: 50 # Hz

      joint_state_broadcaster:
        type: joint_state_broadcaster/JointStateBroadcaster

  joint_state_broadcaster:
    ros__parameters:
      extra_joints: ["right_wheel_joint", "left_wheel_joint"]
      update_rate: 10 # Not Hz! This is 5 Hz, see https://github.com/ros-controls/ros2_control/issues/872
```

```
Node(
    package="controller_manager",
    executable="ros2_control_node",
    parameters=[
            os.path.join(
                get_package_share_directory("amr_bringup"),
                "config/control/amr_controllers.yaml",
            )
    ],
    output="screen",
    remappings=[
        ("~/robot_description", "/robot_description"),
        ("~/cmd_vel", "collision_monitor/cmd_vel"),
        ("~/odom", "odom"),
    ],
    emulate_tty=True,
),
```

I've been running it like this for a long time without issues, that's why I was asking if something recent might have caused this bug. In the coming days I will try to investigate what could be the root cause if you don't have more ideas. Thanks for your initial help!

saikishor, 2023-11-09:
@tonynajjar Sorry I thought it was a different issue. Can you check if the state interfaces exist, by running the command : `ros2 control list_hardware_interfaces` ?