ros-controls/ros2_control #736: What is the expected behavior for missing loop rate?

tylerjw, 2022-06-18:
Currently when the loop rate is missed following iterations "catch up" because the design is to ask for the loop to start at a set time.

If you combine that with trajectory execution and position control the resulting command values are set with close to 0 charge in position.

If your robot takes control messages and the robot always takes the loop rate amount of time in it's internal controller to process a single command you have now have a problem.

You have commanded as close to maxim possible acceleration at your control rate because in one command the robot was moving and in the next n commands while ros control "catches up" are sent with zero position change.

Is there anything we should charge about the way the loop works to make it more robust to position control when run on a non deterministic system like normal Ubuntu?

AndyZe, 2022-06-18:
This is a good reason to get the Ruckig smoothing PR merged:  https://github.com/ros-controls/ros2_controllers/pull/324

You can actually get infinite vels/accels right now, with the spline interpolation algorithm.

AndyZe, 2022-06-18:
I don't completely understand the loop rate issue. Maybe it would be smart to reject incoming commands if `time_since_previous_command` is less than 0.8x the control period? Or something like that.

tylerjw, 2022-06-18:
> I don't completely understand the loop rate issue. Maybe it would be smart to reject incoming commands if `time_since_previous_command` is less than 0.8x the control period? Or something like that.

This is exactly what I did and it greatly improved the performance.

I'll try to draw a diagram or something to make it more clear. I think the current design is reasonable with regard to loop rates but when combined with how the interpolation works and position control it creates this nastily behavior.

tylerjw, 2022-06-18:
[image]

Here is an example of the result on my hardware. The first column is the time delta in calls to my write function, and the rest are the position delta for each joint.

gavanderhoorn, 2022-06-20:
To me, this looks like what I would expect from a non-deterministic system.

Some periods are too short, others too long.

tylerjw, 2022-06-20:
> To me, this looks like what I would expect from a non-deterministic system.
> 
> Some periods are too short, others too long.

I agree, but I think there are things we could do to make this more robust to running on a non-deterministic system. As it is failures are constant and poorly handled.

gavanderhoorn, 2022-06-20:
What do you suggest?

Almost all control systems I am familiar with assume a fixed `dT`. If that's not achieved, behaviour becomes almost 'undefined'.

Dropping "too old commands" is not really an option when you're working with a position controlled system (as the system-under-control would stop), which makes it difficult to come up with a generic solution.

tylerjw, 2022-06-20:
I don't think it is helpful to try to "catch up" in the case of position control because hardware does assume fixed dt and will take that as a hard command to stop motion.

Instead, I think we could calculate the start of the next iteration without trying to catch up by sending commands with near zero dt between them.

tylerjw, 2022-06-20:
Why would control engineers choose this then to write their control systems with if they don't start with the assumption that it isn't going to be deterministic. To my understanding ros2_control does not support any deterministic OS schedulers. It is open source and there is nothing stopping someone from making it work in a deterministic system, but by default it targets normal Linux.

Unless it wants to target vxworks or preempt or some other more deterministic system I think it should try to fail softly when it does not meet the loop rate.

tylerjw, 2022-06-20:
When you say responsibility of the controller do you mean in this case joint trajectory controller? If the controllers responsibility is to see that it is asked to create a new command at a much shorter than normal dt and it fails the current design of ros2_control has no way of communicating that failure to the hardware controller.

tylerjw, 2022-06-20:
If what you mean by controller is the hardware interface, that is what I'm currently doing. However I found this choice to send "catch up" position controls surprising, hence the issue.

gavanderhoorn, 2022-06-20:
> Why would control engineers choose this then to write their control systems with if they don't start with the assumption that it isn't going to be deterministic.

I assume you mean to write: "if they can't start with the assumption it's going to be deterministic".

Control engineers I've talked to and who've used `ros_control` _do_ assume it's capable of providing deterministic behaviour. That's why I started posting comments on your issue: I recognised what you reported.

> To my understanding ros2_control does not support any deterministic OS schedulers.

Separation of concerns would suggest that would also not necessarily be a responsibility of `ros2_control`. There are plenty of other systems capable of scheduling tasks (your OS fi). What would it need to explicitly support?

> It is open source and there is nothing stopping someone from making it work in a deterministic system [..] Unless it wants to target vxworks or preempt or some other more deterministic system

For non-safety-critical systems this would certainly be sufficient in most cases.

By running your controller manager in a thread which is configured with something like FIFO or RR scheduling and an RT priority on a `PREEMPT_RT` kernel (or Xenomai) would already get you much better behaviour, scheduling-wise.

> but by default it targets normal Linux.

I don't know whether this is true. I don't believe it was true for `ros_control`, but perhaps `ros2_control` has changed this.

> When you say responsibility of the controller do you mean in this case joint trajectory controller?

If that would be the active controller, then: yes.

> If the controllers responsibility is to see that it is asked to create a new command at a much shorter than normal dt and it fails the current design of ros2_control has no way of communicating that failure to the hardware controller.

unless we're talking about a RR scheduled hard real-time system, a "shorter dT" is not a failure. IIRC, periods *are* communicated to controllers, via the call to `update(..)`.

Controllers can inspect that argument, and should take the length of the period into account. They probably often do, it's only part of what's going on: scheduling of the rest of your task is also important.

If the reported period to a controller differs from the actual time between `read(..)` and `write(..)`, you'll still end up with the spikes you see in your plot.

> If what you mean by controller is the hardware interface, that is what I'm currently doing. However I found this choice to send "catch up" position controls surprising, hence the issue.

no, a hardware interface's responsibility should be (in my opinion) transforming domain concepts from `ros2_control` to-and-from whatever the targetted hw-system uses, combined with offering communication with that system (and some coordination is probably needed, as there are typically things like control modes which need to be switched, and protocols can be quite complex).

Perhaps if the targetted system offers some way of mitigating missed control periods the hw interface could share some responsibility (as it would probably be in a good position to communicate that), but in principle, it would make sense to me for the active controller to compensate for variable `dT` if possible.

As I wrote: dropping or repeating actuation commands for position controlled actuators will likely result in stuttering. For velocity and force controlled actuators it's less of a problem. That seems like a control decision to me, hence my suggestion to make controllers responsible.

AndyZe, 2022-06-20:
Thank you for the reality check @gavanderhoorn  :+1:

AndyZe, 2022-06-20:
> Why would control engineers choose this then to write their control systems with if they don't start with the assumption that it isn't going to be deterministic.

Some robot drivers seem to handle the jitter in commands much better than others. UR and ABB do a great job. I think you're working with a less sophisticated, raw driver that doesn't handle the jitter well.

Which is to say, we usually don't use a realtime kernel with UR robots (for example) because we can get away with it. But the kernel running on the UR pendant itself is realtime, I believe.

tylerjw, 2022-06-20:
When I said it does not support other schedulers I was simply referring to how the main loop is written and that it does not call any of the scheduler configuration API functions. 

> By running your controller manager in a thread which is configured with something like FIFO or RR scheduling and an RT priority on a PREEMPT_RT kernel (or Xenomai) you would already get much better behaviour,

This is what I mean. ros2_control does not do this in the implemented control loop... There is nothing stopping me or anyone else from doing this themselves though.

I do think the project could though make some decisions to fail in a softer way when run on a non deterministic system if it doesn't out of the box target being run on a system with a scheduler that has options for RR, FIFA, or deadline.

tylerjw, 2022-06-20:
To be clear I've started working on tooling to build ROS for VxWorks because I want to solve this problem for good and make it accessable. The problem though is I just haven't finished that project.

gavanderhoorn, 2022-06-20:
ROS 1 or ROS 2?

Doesn't Wind River already maintain a ROS 2 build for VxWorks 7?

tylerjw, 2022-06-20:
ROS 2... I should have googled before I started that. 😢

gavanderhoorn, 2022-06-20:
Perhaps taking a look at OROCOS could help at this point.

There was a OROCOS compatible `ros_control` `ControllerManager` component available, and OROCOS separates coordination from computation (and the rest of the Cs) *much* better than we have ever done in ROS (1 and 2).

zultron, 2022-06-21:
In my mind, the above discussion divides into two problems.

Relevant to the `ros2_control` code base, one problem is the absence of any RT threading features in the ROS2 [`controller_manager` node][1].  IMO those features would greatly improve the utility of this code base, since users would be a lot closer to acceptable RT performance right out of the box.  Optimizing biggest effect on latency vs. implementation effort, several important features worth adding are `SCHED_FIFO` scheduling, elevated RT scheduling priority, easy granting `CAP_SYS_ADMIN` & `CAP_SYS_NICE` privs for those to either the executable or the user, memory locking to avoid page faults, power management and `/dev/cpu_dma_latency`, and CPU isolation w/cgroups.  Those would make a drastic difference on both `RT_PREEMPT` and vanilla distro kernels.  The [`cyclictest` program][2] implements most of these, and could serve as an example.  Alternatively, run the `controller_manager` in another RT programming environment, such as OROCOS, or (my approach) [Machinekit HAL][3].

[1]:  https://github.com/ros-controls/ros2_control/blob/master/controller_manager/src/ros2_control_node.cpp
[2]:  https://wiki.linuxfoundation.org/realtime/documentation/howto/tools/cyclictest/start
[3]:  https://github.com/tormach/hal_ros_control/tree/foxy-devel

The other problem, where commanded position doesn't account for variation in the update period/`dT`, sounds like an issue in the [`ros2_controllers` repo][4] or @tylerjw 's controller.  The `controller_manager` node passes both the current time and current period (i.e. duration since previous update) to the controllers in its [`update()`][5] method, and it's up to the controllers to do the right thing with that.  The `joint_trajectory_controller` `update()` function [calls][6] its `sample()` method, which in turn [calls][7] its [`interpolate_between_points()` method][8].  The interpolators should use the current time (not the expected update time) as input.  Note that other controllers like the `forward_command_controller` have no concept of a trajectory, and forward whatever command they're told, and it's up to higher-level layers to deal with what command to send for any given update time `T`.

[4]:  https://github.com/ros-controls/ros2_controllers
[5]:  https://github.com/ros-controls/ros2_control/blob/master/controller_manager/src/ros2_control_node.cpp#L63
[6]:  https://github.com/ros-controls/ros2_controllers/blob/master/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L186-L187
[7]:  https://github.com/ros-controls/ros2_controllers/blob/master/joint_trajectory_controller/src/trajectory.cpp#L128
[8]:  https://github.com/ros-controls/ros2_controllers/blob/master/joint_trajectory_controller/src/trajectory.cpp#L152

Also note that nobody has figured out how to (& I suspect never will) achieve perfectly deterministic real time latency in software threads on PC or ARM architectures under a Linux (or Xenomai or any other RT) threading environment(s).  It IS fairly easy to find combinations of hardware, OS and OS settings, and RT programming techniques that limit worst-case latency to under 100µs.  Controllers must be designed for this, as the `joint_trajectory_controller` is.   The best RT configuration I've ever seen was under 8µs max latency, but in my practical experience, there's actually enough jitter in the controller manager `update()` return time (up to 60µs on an Atom CPU recently tested) that further hardware, OS & RT application thread optimization won't help beyond some point and we have to start looking into `ros2_control` & `ros2_controllers`.

destogl, 2022-06-27:
Just for the record at maintainer of the framework: I fully agree with you guys and most of these things actually on my roadmap, but without any timestamp on it, unfortunately.

@tylerjw and @zultron you described some great ideas here, and I appreciate very much. What I would appreciate even more are some practical examples, documentation how to set up this or even PR to the existing code-base. If this means we have to add another version of `controller_mananger` or the `ros2_control_node` for this, no problem, I am open for all the possible solutions.

> The best RT configuration I've ever seen was under 8µs max latency, but in my practical experience, there's actually enough jitter in the controller manager update() return time (up to 60µs on an [...]

You are fully right. Implementation of some (parts) of controllers and framework itself really need some love and deep check what is happening in there w.t.r. of memory allocation and time performance.

destogl, 2022-06-27:
And to answer to the question: “There is no expected behavior. You should take care that you never miss the loop rate.” If you are missing, it more often, try RT_PREEMT or run `ros2_control` on a separate machine. Sorry that I don't have a better solution for now.

Your proposal seems sensible, would you like at least write up a concept of it in the [roadmap](ros-controls/roadmap) repository?

gavanderhoorn, 2022-06-27:
> The best RT configuration I've ever seen was under 8µs max latency, but in my practical experience, there's actually enough jitter in the controller manager `update()` return time (up to 60µs on an [...]

as another 'reality check' perhaps: in my experience, users with a requirement to reach the kind of determinism and low jitter @zultron describes here skip using a "complex OS" like Linux altogether (patched with PREEMPT or Xenomai or not).

Trying to cover those use-cases with vanilla `ros2_control` would perhaps not be the best goal for current development.

I would suggest to set a realistic goal, which covers the majority of use-cases ("diff drive on a 2 wheel + 1 caster" and "6 dof industrial robot with an external motion interface").

Determinism there is important, and low jitter as well, but I doubt control performance in *those* cases would be very much worse if jitter would be a couple hundred micro-seconds.

tylerjw, 2022-06-27:
To be clear, my requirement for jitter for my current piece of hardware is around 1ms.

I am going to work on getting rt_preempt working this week for me to see if I can make that work for my application.

I do think it would be ideal for ros_control to eventually target bare metal for some specific targets using a rtos. Maybe using micro-ros for the external ros interface.

tylerjw, 2022-06-30:
Here is my PR I'm testing with: https://github.com/ros-controls/ros2_control/pull/748

closed 2025-05-26 by github-actions[bot]