ros-controls/ros2_control #759: Add `cleanup_controller` service and method to controller manager

destogl, 2022-07-02:
## Background

Better support for controller's lifecycle, that is the same as for the Lifecycle node. To transition between `inactive` and `unconfigured` state, we should provide service that enables that. The event name is `cleanup`.

Do the following:

1. Add new method cleanup controller in the controller manager.
2. Copy cleanup content form the `unload_controller` method.
3. Add new service called `cleanup_controller` in the CM.
4. Add new CLI for this function, i.e., enable setting in `unconfigured` state controller from CLI.

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/galactic/Tutorials/Workspace/Creating-A-Workspace.html), for Step3 use "Download Source Code" section with [these instructions](https://ros-controls.github.io/control.ros.org/getting_started.html#compiling).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://ros-controls.github.io/control.ros.org/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/galactic/Tutorials.html)
* [ROS Answers](https://answers.ros.org/questions/)

**Good luck with your first issue!**

MOLOCH-dev, 2023-02-02:
Hi, I'd like to take up this issue. Could you please assign it to me @destogl

bailaC, 2023-12-14:
@destogl , I can pick this.

bailaC, 2023-12-21:
https://github.com/ros-controls/ros2_control/pull/1236

soham2560, 2025-04-17:
as discussed in https://github.com/ros-controls/ros2_control/pull/2163#pullrequestreview-2773753961 I would like to pick this up, @bailaC hope thats fine, I'll try to help get your PR to the finish line

christophfroehlich, 2025-07-26:
@soham2560 have you started working on this? Or still waiting for a response [here](https://github.com/ros-controls/ros2_control/pull/1236#issuecomment-2918259172)?

soham2560, 2025-07-26:
Had received a response, just lost track of it. was looking for opinions on https://github.com/ros-controls/ros2_control/pull/1236#issuecomment-2918255540 if possible, not entirely necessary but would be appreciated

will get on this