ros-controls/ros2_control #760: Add `set_controller_state` service to controller manager

destogl, 2022-07-02:
## Background

Service should be equivalent to `set_hardware_compoent_state` so that controller manager takes care of all transitions to the desired state. This service should replace than all the individual services used in CLI.

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/galactic/Tutorials/Workspace/Creating-A-Workspace.html), for Step3 use "Download Source Code" section with [these instructions](https://ros-controls.github.io/control.ros.org/getting_started.html#compiling).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://ros-controls.github.io/control.ros.org/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/galactic/Tutorials.html)
* [ROS Answers](https://answers.ros.org/questions/)

**Good luck with your first issue!**

bmagyar, 2023-05-29:
@destogl is this still relevant?

debanik123, 2024-10-17:
# serice to reload controller -->
ros2 control list_controllers
ros2 service call /controller_manager/switch_controller controller_manager_msgs/srv/SwitchController "{start_controllers: [''], stop_controllers: ['<your_controller_name>'], strictness: 1, start_asap: true}"
ros2 control list_controllers
ros2 service call /controller_manager/unload_controller controller_manager_msgs/srv/UnloadController "{name: '<your_controller_name>'}"
ros2 control list_controllers
ros2 service call /controller_manager/load_controller controller_manager_msgs/srv/LoadController "{name: '<your_controller_name>'}"
ros2 control list_controllers
ros2 service call /controller_manager/configure_controller controller_manager_msgs/srv/ConfigureController "{name: '<your_controller_name>'}"
ros2 control list_controllers
ros2 service call /controller_manager/switch_controller controller_manager_msgs/srv/SwitchController "{start_controllers: ['<your_controller_name>'], stop_controllers: [''], strictness: 1, start_asap: true}"
ros2 control list_controllers  

or, 
ros2 control set_controller_state <your_controller_name> inactive
ros2 control unload_controller <your_controller_name>
ros2 control load_controller <your_controller_name> --set-state active

louietouie, 2025-05-22:
If this is still relevant, I would be interested in attempting it.

One question, is this issue requesting that a new method `list_hardware_components_srv_cb` and corresponding service be added to `controller_manager.cpp`, or just for a new *python* service be added to `controller_manager_services.py` called `set_controller_state`, that makes use of the existing `configure_controller`, `list_controllers`, `switch_controllers`, and upcoming `cleanup_controller` from #759 to switch been unconfigured, active, and inactive?

Either way I'm imagining this would also entail removing the logic from the CLI verb `set_controller_state.py` to use the new service, which seems like it may overlap with #688.

louietouie, 2025-05-22:
Forgot to add mentions... @christophfroehlich @destogl

closed 2025-06-18 by christophfroehlich