ros-controls/ros2_control #764: Can't add description to controller parameters

pac48, 2022-07-06:
The `ControllerInterfaceBase` class sets `automatically_declare_parameters_from_overrides` to true by default. Controllers that inherit from this class have no way to change that default because init is not a virtual method. When parameter are declared automatically using the overrides, a default parameter description is given to them. As far as I know, there is no way to change a parameters description once it has been declared.

It would be nice if the controller could chose whether or not to automatically declare parameter.

bmagyar, 2022-07-07:
Does https://github.com/ros-controls/ros2_control/pull/757 fix this?

pac48, 2022-07-07:
That does fix the issue. Thanks for the quick response!

closed 2022-07-15 by pac48