ros-controls/ros2_control #775: Set the RT thread priority to a default that doesn't require `sudo`

AndyZe, 2022-07-15:
Related to this PR:  https://github.com/ros-controls/ros2_control/pull/748

It would be nice if the thread priority could be set without `sudo`. The Linux manual says this:

On Linux, the RLIMIT_NICE resource limit can be used to define a
       limit to which an unprivileged process's nice value can be
       raised; see [setrlimit(2)](https://man7.org/linux/man-pages/man2/setrlimit.2.html) for details.
       
https://man7.org/linux/man-pages/man7/sched.7.html

I have trouble determining what a good default would be, but I've only looked at it for a few minutes so far.

gavanderhoorn, 2022-07-15:
Would you not want to allow the user starting the process access to the required resources by configuring the `ulimits` instead?

See [this](https://frankaemika.github.io/docs/installation_linux.html#allow-a-user-to-set-real-time-permissions-for-its-processes) for an example.

AndyZe, 2022-07-15:
OK, I'll change the issue title to suggest documenting that ^. Thanks

PR #781 (2022-07-20): Add documentation for realtime permission
Fixes #775

I did test this on actual hardware.

@tylerjw
closed 2022-07-20 by bmagyar