ros-controls/ros2_control #785: Non-default controller_manager node names applied to spawned controllers

shonigmann, 2022-07-26:
**Describe the bug**
Setting the `controller_manager`'s node `name` at launch also overrides the name of any controller nodes spawned by the renamed `controller_manager`, ignoring names specified in either the spawner node or the yaml file. 

This results in multiple nodes with the same name, issues with parameter scope, etc. 

The root cause is probably this issue with rclcpp that's been open for 3 years (https://github.com/ros2/rclcpp/issues/843), but I wanted to document it here in case a) a workaround is possible, or b) to add some visibility in case someone else runs into the same issue down the line. While not essential, in some cases it would be nice to be able to rename nodes rather than namespace the default name.

**To Reproduce**
Steps to reproduce the behavior:
1) in a clean ROS2 galactic workspace clone a fork of the ros2_control_demos galactic branch, which applies a node name to the controller_manager:
```
git clone https://github.com/shonigmann/ros2_control_demos.git -b controller_manager_rename
```
2) install dependencies and build the workspace
3) launch the rrbot demo:
```
ros2 launch ros2_control_demo_bringup rrbot.launch.py
```

**Expected behavior**
The controller manager gets renamed to `renamed_controller_manager`.
The `JointStateBroadcaser` node gets assigned the name `joint_state_broadcaster` defined in rrbot_controllers.yaml.
The `ForwardCommandController` node similarly gets assigned the name `forward_position_controller`.

Introspecting the nodes that are running reflects this, e.g.: 
```
$ ros2 node list
...
renamed_controller_manager
forward_position_controller
joint_state_broadcaster
```

**Actual Behavior**
The controller_manager and controllers are all named `renamed_controller_manager`. 
The `ForwardPositionController` fails to start because the `joints` parameter is undefined (because it looks for the parameter in the scope of the `renamed_controller_manager` rather than `forward_position_controller`).

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Version Galactic

This presumably impacts other distros as well.

sshazly, 2022-07-26:
I was able to reproduce this error, but it seems to be more of an issue with how ros2 python launch system applies the new name to the node that is being launched. https://design.ros2.org/articles/static_remapping.html <-- in this document they talk about renaming a node with subnodes as part of the process. 

The fix that I found will work is to pass another remapping for the ros2_control_node part of the launch process:
from this:
```
    control_node = Node(
        package="controller_manager",
        executable="ros2_control_node",
        name="renamed_controller_manager",
        parameters=[robot_description, robot_controllers],
        remappings=[
            (
                "/forward_position_controller/commands",
                "/position_commands",
            ),
        ],
        output={
            "stdout": "screen",
            "stderr": "screen",
        },
    )
```

to this:

```
    control_node = Node(
        package="controller_manager",
        executable="ros2_control_node",
        parameters=[robot_description, robot_controllers],
        remappings=[
            (
                "/forward_position_controller/commands",
                "/position_commands",
            ),
            (
                "controller_manager:__name",
                "renamed_controller_manager"
            ),
        ],
        output={
            "stdout": "screen",
            "stderr": "screen",
        },
    )
```

A little frustrating that the name argument does not work as advertised in the ros2 launch system

shonigmann, 2022-07-26:
this does solve the issue on my end as well, cheers!

closed 2022-07-26 by shonigmann