ros-controls/ros2_control #793: What is the suggested way to handle hardware intefaces running at different rates

VladimirIvan, 2022-08-02:
I have a few HW interfaces that run at higher rate (200Hz) and one interface that runs at a lower rate (30Hz). The slower interface does use blocking calls when performing `read()` and `write()` so it would limit high rate interfaces to 30Hz when called in the same control loop. I do need to run some controllers that talk to both the fast and slow interfaces at the same time so I can't run each interface in a separate node.

My solution would be to make the slow interface run in separate thread and synchronise it with the main loop. Commands will be picked up at 30Hz and states will be repeated at 200Hz until a new `read()` updates them in the parallel thread. This will not need any new features in ros2_control so it's not a feature request but I'm wondering if this is the right solution. Are there any features already implemented that would solve this issue that I may have missed in the documentation?

Thanks

bmagyar, 2022-08-02:
Hey @VladimirIvan indeed that is the currently recommended solution. We are (specifically @VX792  :D ) working on adding support for this exact thing but provided by the framework.

VladimirIvan, 2022-08-02:
Great! I don't have a better idea and I'm happy test your version too.
One thing worth considering is how the different rates will affect controllers. Especially if for example velocities are calculated using numdiff two repeated position states will mess this up. I'm guessing you would have encountered similar issues as with async controllers so whatever worked there should work here.

Keep up the good work then and let me know when you have something for me to test.

saikishor, 2025-04-08:
This is already solved with https://github.com/ros-controls/ros2_control/pull/1570.
I'll go ahead and close this. 

Thank you

closed 2025-04-08 by saikishor