ros-controls/ros2_control #796: Add `list_hardware_components` CLI

destogl, 2022-08-04:
## Background

Add CLI command for `list_hardware_components` service and make nice colored output of it.

See for example how `list_hardware_interfaces` CLI is done.

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/galactic/Tutorials/Workspace/Creating-A-Workspace.html), for Step3 use "Download Source Code" section with [these instructions](https://ros-controls.github.io/control.ros.org/getting_started.html#compiling).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://ros-controls.github.io/control.ros.org/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/galactic/Tutorials.html)
* [ROS Answers](https://answers.ros.org/questions/)

**Good luck with your first issue!**

Serafadam, 2022-08-09:
Hey, a week ago I posted on [colcon-core](https://github.com/colcon/colcon-core/issues/523) repository with an idea to use [rich](https://github.com/Textualize/rich) library from python to improve readability of colcon's output, maybe this could also be applied here?

mcevoyandy, 2023-01-09:
@destogl I was going to try and tackle this issue, but am running into some commit problems I can't seem to fix:

1. On a new python file, ament_copyright fails when copying the copyright from an existing file and returns with `license=<unknown>`
2. `pep257` returns import error `cannot import 'Set' from 'collections'`

how do I resolve these to pass the pre commit hook?

mcevoyandy, 2023-01-09:
How should this message be printed out?

Using `rrbot.launch.py` as an example should formatting be something like this?

```
Hardware Component 0
	name: RRBot
	type: system
	plugin name: ros2_control_demo_hardware/RRBotSystemPositionOnlyHardware
	state: id=3 label=active
	command interfaces
		joint1/position [available] [unclaimed]
		joint2/position [available] [unclaimed]
	state interfaces
		joint1/position [available] [unclaimed]
		joint2/position [available] [unclaimed]
```

For `state`, is there a better format or coloring to tie it to available states [here](https://github.com/ros2/rcl_interfaces/blob/humble/lifecycle_msgs/msg/State.msg)? 
What are the cases where `ListHardwareComponents.srv` returns more than one component?

bmagyar, 2023-01-10:
I like the above example!
Also, the cases of multiple components returned are when you have a robot made up of multiple self-contained components, the canonical one being a robot arm attached to a mobile base.

PR #891 (2023-02-08): Closes #796 - Adds list_hardware_components to CLI
Adds CLI for `list_hardware_components`. Closes #796.
commit e7bab63b (2023-02-08): Closes #796 - Adds list_hardware_components to CLI (#891)
commit 61538d55 (2023-02-14): Closes #796 - Adds list_hardware_components to CLI (#891) (#937)

(cherry picked from commit e7bab63b2cd7bd2416ac270f5dab8049afcdd656)
closed 2023-02-08 by bmagyar