ros-controls/ros2_control #798: Transmission interface

Cryoschrome, 2022-08-05:
Hi, 
I have a stepper motor with 50:1 gear ratio that I am trying to move through MoveIt2! and ros2_control. Due to which I am getting the PATH_TOLERANCE_VIOLATION.  I am guessing to tell the controller about the gear ratio I need to utilize the transmission interface. If so, could you advice me on where I can find instructions to implement it.
I am using Ubuntu 22.04 and ROS humble.

Cryoschrome, 2022-08-10:
Can someone please provide support on how to implement transmission interface?

destogl, 2022-10-02:
Transmission interface has to be implemented on hardware side not on the controller side. To be able to help you, you should provide many more details. From your description, I have no idea what is your issue.

destogl, 2022-11-17:
Closing this because it is rather set up issue without activity. @Cryoschrome feel free to reopen.

closed 2022-11-17 by destogl