ros-controls/ros2_control #800: Wrong package arrangment

Llavrih, 2022-08-11:
**Describe the bug**
I am including #include <controller_manager/controller_manager.hpp> in my cpp program. When trying to build the pkg 
I get the error: 
```fatal error: controller_manager_msgs/srv/configure_controller.hpp: No such file or directory```.
As I checked there is a problem with package arrangment in /opt/ros/humble/include/controller_manager_msg

```
lenartl@robot2204:/opt/ros/humble/include/controller_manager_msgs$ ls
controller_manager_msgs
lenartl@robot2204:/opt/ros/humble/include/controller_manager_msgs$ ls controller_manager_msgs/
msg  srv
```

**Environment**
 - OS: Ubuntu 22.04
 - Version Humble

gavanderhoorn, 2022-08-21:
I believe this is actually as-intended. Refer to https://github.com/ros2/ros2_documentation/pull/2915 for the update to the Humble documentation.

destogl, 2022-10-02:
This arrangement is correct, what maybe is not correct that we are not exporting properly dependent packages.

OK, [this](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/CMakeLists.txt#L194) and [this](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/CMakeLists.txt#L11) line show that we are exporting them correctly.

Did you solve the issue?

Since I don't see a bug here I'll close this. You are welcome to reopen if you think that is necessary.

closed 2022-10-02 by destogl