ros-controls/ros2_control #807: How to delete controller manager correctly?

GilmarCorreia, 2022-09-01:
Hello ROS developers,

I am developing an application that needs both configure and delete ROS2 controllers. I am trying to figure out the best way to delete a controller manager node. Which is the best practice?

Creation methods:
```cpp
Robot *currentRobot =new Robot();
std::string controllerManagerNodeName = currentRobot->getCM_Nodename();

auto *cm = new controller_manager::ControllerManager(std::move(resource_manager_ptr),this->simRos2ControlPtr->executor_, controllerManagerNodeName);

currentRobot->controller_manager_.reset(cm);

executor_->add_node(currentRobot->controller_manager_);

for (auto controller : currentRobot->controllers_){
    currentRobot->controller_manager_->load_controller(controller->name,  controller->type);
    currentRobot->controller_manager_->configure_controller(controller->name);
}
```

deletion methods:
```cpp
for (auto controller : currentRobot->controllers_){
    currentRobot->controller_manager_->unload_controller(controller->name);
    std::this_thread::sleep_for(std::chrono::milliseconds(100));
}

executor_->remove_node(currentRobot->controller_manager_);
currentRobot->controller_manager_->~ControllerManager();
```

But, If I try to delete this error message shows up: 

```bash
Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
         at line 127 in /tmp/binarydeb/ros-foxy-class-loader-2.0.2/src/class_loader.cpp
```

Is that the correct way to do that? Thank you.

**Environment:**
 - OS: Ubuntu 20.04 (wsl2)
 - Version: Foxy

destogl, 2022-09-05:
> I am developing an application that needs both configure and delete ROS2 controllers. I am trying to figure out the best way to delete a controller manager node. Which is the best practice?

I understand that you want to deactivate and unload controllers, but why do you want to delete controller manager? That is rather unusual use-case.

> But, If I try to delete this error message shows up:
> 
> ```shell
> Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
>          at line 127 in /tmp/binarydeb/ros-foxy-class-loader-2.0.2/src/class_loader.cpp
> ```

This error message is probably caused by the resource manager that is also loaded. So before deleting controller manager, you should also deactivate and unload hardware components. To set hardware state it's easy and there is interface for it, but for unloading hardware I am not sure we have something that support this at the moment.

For this, you should first add this interface to the Resource Manger to be able to unload hardware components and then also a method in the controller manager. And this method has to be called in the destructor of the controller manager and resource manager.

>     * Version: Foxy

If you have any possibly, I would recommend you to use master branch (ROS rolling or humble) because Foxy is ancient and many new features are missing there. Moreover, it could be quite hard to get this functionality into Foxy since many, many things are not restructured to be flexible and reusable.

bmagyar, 2022-11-15:
Calling the constructor by hand in C++ is always a red flag. Closing this for now due to lack of clarity and activity. 
Feel free to reopen.

closed 2022-11-15 by bmagyar