ros-controls/ros2_control #817: initial_value params for state interfaces not respected

Yadunund, 2022-09-23:
## Background
Hi all,

My setup is as follows:
* Ubuntu 22.04
* ROS 2 Humble installed via binaries
* ros2_control and ros2_controllers installed via binaries

I've been using the `JointTrajectoryController` to command a couple of ABB industrial robots. I'm using [this](https://github.com/PickNikRobotics/abb_ros2) hardware interface implementation. I notice that once the controller starts and the command interfaces are claimed (both position and velocity), the robot always gets commanded to reach `0.0` joint value for all its joints. This is despite explicitly specifying the `initial_value` params for each position and velocity state interface in the `ros_control.xacro` file.

```
        <state_interface name="position">
          <param name="initial_value">-0.512</param>
        </state_interface>
``` 

My expectation is that once the interfaces are claimed by the controller, the robot will continue to remain idle in it current joint configuration (as specified by the `initial_value` param) and not move to 0.0 joint values.

Is this happening due to an error I might have made in my `ros2_control.xacro` or `controllers.yaml` files? Or could it be due to the implementation of the hardware_interface itself? Any insights would greatly help me in debugging this problem.

Thanks in advance!

destogl, 2022-10-01:
> I've been using the `JointTrajectoryController` to command a couple of ABB industrial robots. I'm using [this](https://github.com/PickNikRobotics/abb_ros2) hardware interface implementation. I [...]

This is definitely not the behavior you want to have. So, what happens is that JTC gets the initial values from hardware interface after it is started. If those values are wrong, then this can happen.
Please do the following to debug this:

1. Which version of JTC/ros2_control are you using?
2. what are the values on `/joint_states` topic before you activate JTC? Are they 0? If yes, then HW interface is not reporting correct values.
3. If they are correct, then you are maybe using too old version of JTC. (If you are using Foxy, this can be an issue.)

Yadunund, 2022-10-11:
Hi @destogl,

Many thanks for the reply and apologies for the late follow up, I wasn't subscribed to repo notifications and just enabled that. 

1. I'm using the latest humble binaries

2. I will double check this by tomorrow and get back to you. I suspect this is what is happening.

destogl, 2022-11-17:
@Yadunund and news?

closed 2022-11-17 by destogl

Yadunund, 2022-11-18:
Thanks @destogl, I think this is most likely an issue with the implemented hardware_interface.

Yadunund, 2022-11-18:
The issue is indeed with the `hardware_interface` implementation. I've documented the problem with the implementation here https://github.com/PickNikRobotics/abb_ros2/issues/44. The `initial_value` param in the ros2 control xacro works as expected when the `hardware_interface` is working properly 👍🏼