ros-controls/ros2_control #825: test_load_controller tests passing regardless of missing controller plugin

ijnek, 2022-09-29:
**Describe the bug**
I'm trying to add a new controller, and following [Writing a new controller](https://control.ros.org/master/doc/ros2_controllers/doc/writing_new_controller.html), and wrote the ``test_load_*`` test before exporting the controller as a plugin (following TDD).
Now, my plugin should fail the test because I haven't exported it as a plugin, but the test_load_* tests pass regardless. I haven't looked into why, but I'm guessing ``cm.load_controller`` doesn't throw an exception even if the controller isn't found.

``test_load_ackermann_controller.cpp``:

```cpp
TEST(TestLoadAckermannController, load_controller)
{
  rclcpp::init(0, nullptr);

  std::shared_ptr<rclcpp::Executor> executor =
    std::make_shared<rclcpp::executors::SingleThreadedExecutor>();

  controller_manager::ControllerManager cm(
    std::make_unique<hardware_interface::ResourceManager>(
      ros2_control_test_assets::minimal_robot_urdf),
    executor, "test_controller_manager");

  ASSERT_NO_THROW(
    cm.load_controller("test_ackermann_controller", "ackermann_controller/AckermannController"));

  rclcpp::shutdown();
}
```

Test output:

```sh
test 1
    Start 1: test_load_ackermann_controller

1: Test command: /usr/bin/python3.10 "-u" "/opt/ros/rolling/share/ament_cmake_test/cmake/run_test.py" "/home/ijnek/tmp_workspaces/quasor_ws/build/ackermann_controller/test_results/ackermann_controller/test_load_ackermann_controller.gtest.xml" "--package-name" "ackermann_controller" "--output-file" "/home/ijnek/tmp_workspaces/quasor_ws/build/ackermann_controller/ament_cmake_gmock/test_load_ackermann_controller.txt" "--command" "/home/ijnek/tmp_workspaces/quasor_ws/build/ackermann_controller/test_load_ackermann_controller" "--gtest_output=xml:/home/ijnek/tmp_workspaces/quasor_ws/build/ackermann_controller/test_results/ackermann_controller/test_load_ackermann_controller.gtest.xml"
1: Test timeout computed to be: 60
1: -- run_test.py: invoking following command in '/home/ijnek/tmp_workspaces/quasor_ws/build/ackermann_controller':
1:  - /home/ijnek/tmp_workspaces/quasor_ws/build/ackermann_controller/test_load_ackermann_controller --gtest_output=xml:/home/ijnek/tmp_workspaces/quasor_ws/build/ackermann_controller/test_results/ackermann_controller/test_load_ackermann_controller.gtest.xml
1: Running main() from gmock_main.cc
1: [==========] Running 1 test from 1 test suite.
1: [----------] Global test environment set-up.
1: [----------] 1 test from TestLoadAckermannController
1: [ RUN      ] TestLoadAckermannController.load_controller
1: [INFO] [1664446283.591663194] [resource_manager]: Loading hardware 'TestActuatorHardware' 
1: [INFO] [1664446283.592020831] [resource_manager]: Initialize hardware 'TestActuatorHardware' 
1: [INFO] [1664446283.592039653] [resource_manager]: Successful initialization of hardware 'TestActuatorHardware'
1: [INFO] [1664446283.592054975] [resource_manager]: Loading hardware 'TestSensorHardware' 
1: [INFO] [1664446283.592121221] [resource_manager]: Initialize hardware 'TestSensorHardware' 
1: [INFO] [1664446283.592135944] [resource_manager]: Successful initialization of hardware 'TestSensorHardware'
1: [INFO] [1664446283.592144047] [resource_manager]: Loading hardware 'TestSystemHardware' 
1: [INFO] [1664446283.592216189] [resource_manager]: Initialize hardware 'TestSystemHardware' 
1: [INFO] [1664446283.592230077] [resource_manager]: Successful initialization of hardware 'TestSystemHardware'
1: [INFO] [1664446283.605684095] [test_controller_manager]: Loading controller 'test_ackermann_controller'
1: [ERROR] [1664446283.605767711] [test_controller_manager]: Loader for controller 'test_ackermann_controller' (type 'ackermann_controller/AckermannController') not found.
1: [INFO] [1664446283.605780508] [test_controller_manager]: Available classes:
1: [INFO] [1664446283.605792392] [test_controller_manager]:   controller_manager/test_controller
1: [INFO] [1664446283.605797803] [test_controller_manager]:   controller_manager/test_controller_failed_init
1: [INFO] [1664446283.605802511] [test_controller_manager]:   controller_manager/test_controller_with_interfaces
1: [INFO] [1664446283.605807891] [test_controller_manager]:   diff_drive_controller/DiffDriveController
1: [INFO] [1664446283.605812762] [test_controller_manager]:   effort_controllers/JointGroupEffortController
1: [INFO] [1664446283.605816913] [test_controller_manager]:   force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster
1: [INFO] [1664446283.605822276] [test_controller_manager]:   forward_command_controller/ForwardCommandController
1: [INFO] [1664446283.605827011] [test_controller_manager]:   forward_command_controller/MultiInterfaceForwardCommandController
1: [INFO] [1664446283.605832215] [test_controller_manager]:   imu_sensor_broadcaster/IMUSensorBroadcaster
1: [INFO] [1664446283.605836700] [test_controller_manager]:   joint_state_broadcaster/JointStateBroadcaster
1: [INFO] [1664446283.605840791] [test_controller_manager]:   joint_trajectory_controller/JointTrajectoryController
1: [INFO] [1664446283.605846401] [test_controller_manager]:   position_controllers/JointGroupPositionController
1: [INFO] [1664446283.605850457] [test_controller_manager]:   tricycle_controller/TricycleController
1: [INFO] [1664446283.605854620] [test_controller_manager]:   velocity_controllers/JointGroupVelocityController
1: [INFO] [1664446283.605861582] [test_controller_manager]:   controller_manager/test_chainable_controller
1: [       OK ] TestLoadAckermannController.load_controller (22 ms)
1: [----------] 1 test from TestLoadAckermannController (22 ms total)
1: 
1: [----------] Global test environment tear-down
1: [==========] 1 test from 1 test suite ran. (22 ms total)
1: [  PASSED  ] 1 test.
1: -- run_test.py: return code 0                   
1: -- run_test.py: inject classname prefix into gtest result file '/home/ijnek/tmp_workspaces/quasor_ws/build/ackermann_controller/test_results/ackermann_controller/test_load_ackermann_controller.gtest.xml'
1: -- run_test.py: verify result file '/home/ijnek/tmp_workspaces/quasor_ws/build/ackermann_controller/test_results/ackermann_controller/test_load_ackermann_controller.gtest.xml'
```

**Expected behavior**
Test fails because the controller is missing

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Version: Rolling

**Additional context**
Add any other context about the problem here, especially include any modifications to ros2_control that relate to this issue.

destogl, 2022-10-01:
Are you sure that the controller was not exported at some time before?
I have rather different experience that we get false positives than false negatives on this test.

destogl, 2022-10-01:
I'll close this because it is unclear if there is really an issue in our library or your code. You are welcome to continue commenting and reopen when you are certain that there is an issue in controller manager.

bmagyar, 2023-03-15:
@destogl @ijnek does this issue still persist?

ijnek, 2023-03-30:
> @destogl @ijnek does this issue still persist?

Sorry got your message, but haven't found the time to test this out again. I'll try and give an update this week.

closed 2023-04-29 by bmagyar