ros-controls/ros2_control #832: Segfault when using ROS topics

vectorcrumb, 2022-10-03:
**Describe the bug**
I'm working on porting a hardware interface from ROS2 Foxy to Humble. The interface is used to communicate over ROS topics with ros2_socketcan to control four motors on the CAN network. However, when calling `publish` to send a CAN control package on a LifecyclePublisher object within a LifecycleNode belonging to a `hardware_interface::SystemInterface` object, I get a segfault. This also occurs during operations like `get_subscription_count()` and when trying to create a subscriber. 

**Expected behavior**
Based on issue #740, I understand that using ROS topics with a hardware_interface plugin is strongly discouraged. However, seeing as this code was working fine on Foxy, I wanted to know if any changes were implemented in Galactic or Humble that broke using ROS topics with publishers and subscribers, or if this is a bug I have introduced while porting the code.

I've looked over the source and have found that controllers (such as the [JointTrajectoryController](https://github.com/ros-controls/ros2_controllers/blob/master/joint_trajectory_controller/src/joint_trajectory_controller.cpp)) are able to create LifecycleNodes as members, however I have found no mention of nodes in hardware_interfaces when I was looking over the ros2_control_demos repo.

I've already rewritten the code using the underlying ros2_socketcan classes to directly talk with the CAN interface, however I'm still curious about this issue and would appreciate any insights.

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Version: Humble
 - Occurs on both Raspberry Pi with a real CAN network as well as on my laptop with a virtual CAN network.

destogl, 2022-10-05:
There are many changes in how nodes are handled between Foxy and Humble. Our main change is a switch to LIfecycleNode that handles publishers a bit differently. Still, this doesn't explain segfaults.

Since this usage is *strongly discouraged*. And we don't have actual access to the code, I don't know how I can help you (except you want to professional support, but this is not the right channel for that). Just few comments:

1. Please don't do that.
2. Hardware interface are not nodes, and they don't implement Lifecycle Nodes, they are just having the same lifecycle as the LifecycleNodes.
3. You should always assume that is somewhere in issue in your code before you label something as `bug`.
4. Please take care that when you describe something as a bug, it should be explained in such a way that anyone can easily replicate it. (I know this i hard)

closed 2022-10-05 by destogl

vectorcrumb, 2022-10-05:
Hi, thanks for the answer. And I'm really sorry about the `bug` label, I should've removed it before posting, because as you say, it is most likely an issue with my implementation.

So in summary, hardware interfaces don't have the same capabilities as a LifecycleNode, they simply use that interface to interact with ros2_control. Creating a LifecycleNode subclass member inside a hardware interface and publishing from inside it shouldn't segfault, but regardless, I still shouldn't interacting with ROS from inside the interface.