ros-controls/ros2_control #847: time out ros2_control_node if delayed in start

bmagyar, 2022-10-23:
Discussion opener:

bmagyar, 2022-10-24:
What do you think @destogl ?

destogl, 2022-11-17:
We should actually do the following:

1. Load URDF at start (as we are doing)
2. Don't start hardware components automatically at all and never (this is already a feature)
3. Create “spawners” for hardware components so that the loading and configuration of hardware (when communication should be created at latest) is in a separate thread than the main loop.

Advantages of this approach:

1. CM will get leaner since it will not have business logic of starting hardware anymore.
2. The expected behavior for controllers and hardware is identical.
3. No hardware can block CM if something gets wrong on the loading and configuration (much less crashes of controller manager)