ros-controls/ros2_control #850: Update rate must be an integer?

danzimmerman, 2022-11-05:
**Describe the bug**
In a simple controller package using `generate_parameter_library`, using a floating-point number in an `update_rate.yaml` file throws an error that the update rate param must be an integer.

```
(humble) dan@computer:~/ros/minimal_ws$ ros2 launch dz_minimal_controller controller_load.launch.py
[INFO] [launch]: All log files can be found below /home/dan/.ros/log/2022-11-05-12-08-27-470583-schelkunoff-191619
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [191620]
[INFO] [spawner-2]: process started with pid [191622]
[ros2_control_node-1] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterTypeException'
[ros2_control_node-1]   what():  parameter 'update_rate' has invalid type: expected [integer] got [double]
[spawner-2] [INFO] [1667664507.716583768] [spawner_dz_controller]: Waiting for '/controller_manager' node to exist
[ERROR] [ros2_control_node-1]: process has died [pid 191620, exit code -6, cmd '/home/dan/.local/opt/mambaforge/envs/humble/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_pkemfwt8 --params-file /home/dan/ros/minimal_ws/install/share/dz_minimal_controller/config/update_rate.yaml --params-file /home/dan/ros/minimal_ws/install/share/dz_minimal_controller/config/controller_params.yaml'].
[spawner-2] [INFO] [1667664509.729789056] [spawner_dz_controller]: Waiting for '/controller_manager' node to exist
```

**To Reproduce**
Steps to reproduce the behavior:
 * Possibly can be triggered in other packages by finding the update rate YAML file and changing the rate to a floating-point value
 * Otherwise, the `dz/fp-update-rate` branch here reproduces it with a simple controller: https://github.com/danzimmerman/dz_minimal_controller/tree/dz/fp-update-rate

**Expected behavior**
The update rate in Hertz is can take non-integer values per the documentation:

https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html#parameters

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04 Focal
 - Version: Humble, installed via [Robostack](https://robostack.github.io/)
- Anything that may be unusual about your environment:
 - Mixed 2.12 and 2.13 release binaries because of the way Robostack updates packages.

**Additional context**

Possibly this is user error on my package level. Is there something about defaults in [`generate_parameter_library`](https://github.com/PickNikRobotics/generate_parameter_library/)? 

Maybe this is more of an issue for that repo? I've been working to try to use `ros2_control` properly with `generate_parameter_library` by looking at Joint Trajectory Controller and Admittance Controller, maybe there's something I'm missing.

danzimmerman, 2022-11-05:
Works just fine with an integer update rate parameter:

```
(humble) dan@computer:~/ros/minimal_ws$ ros2 launch dz_minimal_controller controller_load.launch.py
[INFO] [launch]: All log files can be found below /home/dan/.ros/log/2022-11-05-12-26-23-745186-computer-193411
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [193412]
[INFO] [spawner-2]: process started with pid [193414]
[ros2_control_node-1] [INFO] [1667665583.926085302] [resource_manager]: Loading hardware 'ur'
[ros2_control_node-1] [INFO] [1667665583.926278715] [resource_manager]: Initialize hardware 'ur'
[ros2_control_node-1] [WARN] [1667665583.926297342] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example:
[ros2_control_node-1] <state_interface name="velocity">
[ros2_control_node-1]   <param name="initial_value">0.0</param>
[ros2_control_node-1] </state_interface>
[ros2_control_node-1] [INFO] [1667665583.926325110] [resource_manager]: Successful initialization of hardware 'ur'
[ros2_control_node-1] [INFO] [1667665583.926399353] [resource_manager]: 'configure' hardware 'ur'
[ros2_control_node-1] [INFO] [1667665583.926402476] [resource_manager]: Successful 'configure' of hardware 'ur'
[ros2_control_node-1] [INFO] [1667665583.926417945] [resource_manager]: 'activate' hardware 'ur'
[ros2_control_node-1] [INFO] [1667665583.926420426] [resource_manager]: Successful 'activate' of hardware 'ur'
[ros2_control_node-1] [INFO] [1667665583.927869017] [controller_manager]: update rate is 500 Hz
[ros2_control_node-1] [INFO] [1667665583.989580428] [controller_manager]: Loading controller 'dz_controller'
[spawner-2] [INFO] [1667665584.002470027] [spawner_dz_controller]: Loaded dz_controller
[ros2_control_node-1] [INFO] [1667665584.002938087] [controller_manager]: Configuring controller 'dz_controller'
[ros2_control_node-1] [INFO] [1667665584.002972677] [dz_controller]: on_configure() returns SUCCESS
[ros2_control_node-1] [INFO] [1667665584.004058874] [dz_controller]: on_activate() returns SUCCESS
[ros2_control_node-1] [INFO] [1667665584.006032091] [dz_controller]: update() returns OK, printing this once (maybe?)
[spawner-2] [INFO] [1667665584.006484940] [spawner_dz_controller]: Configured and activated dz_controller
[INFO] [spawner-2]: process has finished cleanly [pid 193414]
```

bmagyar, 2022-11-15:
hey, 

I took a quick look and it seems that [this](https://github.com/ros-controls/ros2_control/blob/1ceb627c36725b91c8141808fb231a15e9a4b3c0/controller_interface/src/controller_interface_base.cpp#L36) happened at some point. This probably overrides the one from `generate_parameters_library`?

danzimmerman, 2022-11-16:
Yeah, changing that definitely isn't enough, and now that I poke around a little more it seems like the update rate just needs to be an integer throughout.

https://github.com/ros-controls/ros2_control/search?q=update_rate_

In `controller_interface_base.cpp` it's also treated as an integer [here](https://github.com/ros-controls/ros2_control/blob/1ceb627c36725b91c8141808fb231a15e9a4b3c0/controller_interface/src/controller_interface_base.cpp#L86) and [here](https://github.com/ros-controls/ros2_control/blob/1ceb627c36725b91c8141808fb231a15e9a4b3c0/controller_interface/src/controller_interface_base.cpp#L129) 

I think the docs are just incorrect.

I'll change my parameter definition YAML to validate it as an integer. Doesn't actually matter much for my use case, but wanted to check to see if this was intended.

PR #858 (2022-11-17): [Doc] Correct type of update_rate parameter
I don't see value in having it to be double. Never saw its use as double. So here I a correction of the docs.

closes #850
closed 2022-11-17 by destogl