ros-controls/ros2_control #851: The diagnostic_updater locks controllers with a mutex

AndyZe, 2022-11-07:
As added in this PR:  https://github.com/ros-controls/ros2_control/pull/820

```
  // lock controllers
  std::lock_guard<std::recursive_mutex> guard(rt_controllers_wrapper_.controllers_lock_);
```

This is likely to have a negative effect on realtime performance.

destogl, 2022-11-07:
I don't understand the issue here. The mentioned statement is used in all callbacks since it is the way to fix the controllers lists to not change when reading data from non-realtime loop. This only disables loading of new controllers ([see this line](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/src/controller_manager.cpp#L963)) which is also a procedure not relevant for the realtime.

AndyZe, 2022-11-07:
OK, good. It's not clear from the comment, without knowing exactly what it does beforehand (like you do). Obviously locking controllers sounds dangerous

closed 2022-11-07 by AndyZe

destogl, 2022-11-08:
No problem, thanks for asking. It does not lock controllers, but the lists where those are stored.