ros-controls/ros2_control #872: Individual controller's update_rate is confusing

tonynajjar, 2022-12-06:
Setting an individual controller update rate is confusing. I noticed, according to this [line](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/src/controller_manager.cpp#L1761) that:

real controller update rate (Hz) = controller_manager.update_rate / controller_name.update_rate

```
  controller_manager:
    ros__parameters:
      update_rate: 50 # Hz

      tricycle_controller:
        type: tricycle_controller/TricycleController

      joint_state_broadcaster:
        type: joint_state_broadcaster/JointStateBroadcaster

  joint_state_broadcaster:
    ros__parameters:
      update_rate: 10 # Not the real rate! With this configuration /joint_states is published at 50/10 = 5 Hz
```

saikishor, 2023-07-05:
Hello!

I think the logic has been changed to the counters in the recent versions of ros2_control. However, you should see it as 50 update cycle lead to 1 sec, then the factor 5 that you mentioned above leads to 10 update cycles which is 10Hz. I believe it is right. If you have more questions, feel free to write back.

gwkim-dev, 2023-07-19:
Hi.
I believe there is an issue where the values of update rate and period do not match.

For now, as explained above, setting the parameter to 5 will result in a control rate of 10 Hz. 
But the period argument in the controller's update function will then have a 0.2 seconds. 
I think it should be 0.1 seconds, not 0.2 seconds.

If the update rate of of individual controller refers to the update frequency per second, 
then the parameter should be set to 10 to achieve a 10Hz frequency. 
However, if it represents a ratio for the controller manager's update rate, 
then it can remain as it is, but the method for calculating the period needs to be changed.

bmagyar, 2023-07-19:
Yes, the method of computing went south at some point...
I agree that it needs fixing. Perhaps the best way is to refactor this bit
of code with modulo operators into a function (however trivial) which can
be unit tested should help maintenance

saikishor, 2023-07-19:
> For now, as explained above, setting the parameter to 5 will result in a control rate of 10 Hz.
> But the period argument in the controller's update function will then have a 0.2 seconds.
> I think it should be 0.1 seconds, not 0.2 seconds.

Hello @gwkim0!

Yes, you are right. I see your point. Sorry for the confusion in my earlier message. There seems to be an issue with this condition check, it needs to be updated also considering the controller's update rate into account to have the correct update rate:
https://github.com/ros-controls/ros2_control/blob/6cd8191550b59fcf940227583168dab9674adb13/controller_manager/src/controller_manager.cpp#L1892

I can try to fix it sometime during this week.

closed 2023-09-06 by destogl