ros-controls/ros2_control #876: Custom mock component (e.g. for multi axis servo controller)

firesurfer, 2022-12-09:
Hi I was looking into using ros2control for a setup where simply mirroring certain joints won't be enough to provide a fake hardware state. 

For example I have a servo controller that takes G-Code commands for multiple axis and executes those. In order to mimic that behaviour I want to use the same trajectory  the servo controller would generate.

Therefore I was looking how to add a custom mock component but couldn't find any examples. 
The [documentation page](https://control.ros.org/master/doc/ros2_control/hardware_interface/doc/mock_components_userdoc.html) is also kind of unclear whether this is even possible.

Thanks a lot in advance!

firesurfer, 2022-12-09:
After reading some code I figured it out on my own.  A mock component is basically a normal component but  implements some simply fake hardware. 
Didn't get that from the documentation

closed 2022-12-21 by firesurfer

bmagyar, 2022-12-22:
Hey Lennart,

Would you mind adjusting the docs in this case? Or at least point me in the
right direction please? We'd like to make sure our docs are clear and
useful and not just a bunch of words:)

firesurfer, 2022-12-22:
Hi thanks for your answer. 

Unfortunately I probably wont find the time for creating a PR that adapts the documentation.

When reading the documentation page about mock components the difference between the generic mock component (created via `<plugin>mock_components/GenericSystem</plugin>` ) and the general case (which is basically: Yeah you can have mock components which emulate hardware in a certain way) didn't become clear. 

It would be great if this difference would be emphasized in the documentation as well as perhaps some works about implementing custom mock components (e.g. Implementing your own mock component is as simple as implementing your own hardware component. Instead of interacting with the hardware you provide access to some kind of fake or simulated system representing your hardware component. For more details see the instructions check the instructions on how to implement a hardware component)