ros-controls/ros2_control #880: my robot collision doesn't work when add control plugin

Zarqu0n, 2022-12-17:
**Describe the bug**
My robot gazebo_control.xacro:
```
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">
    <xacro:macro name="gazebo_control">

        <ros2_control name="GazeboSystem" type="system">
            <hardware>
                <plugin>gazebo_ros2_control/GazeboSystem</plugin>
            </hardware> 

            <joint name="haa_joint_6">
                <command_interface name="position">
                    <param name="min">-3.8</param>
                    <param name="max">-0.707</param>
                </command_interface>
                <state_interface name="position"/>
                <state_interface name="velocity"/>
                <state_interface name="effort"/>
            </joint>

            <joint name="hfe_joint_6">
                <command_interface name="position">
                    <param name="min">0.3</param>
                    <param name="max">1.4</param>
                </command_interface>
                <state_interface name="position"/>
                <state_interface name="velocity"/>
                <state_interface name="effort"/>
            </joint>

            <joint name="kfe_joint_6">
                <command_interface name="position">
                    <param name="min">-2.27</param>
                    <param name="max">-1.57</param>
                </command_interface>
                <state_interface name="position"/>
                <state_interface name="velocity"/>
                <state_interface name="effort"/>
            </joint>
        </ros2_control>

        <gazebo>
            <plugin filename="libgazebo_ros2_control.so" name="gazebo_ros2_control">
                <robot_sim_type>gazebo_ros2_control/GazeboSystem</robot_sim_type>
                <parameters>$(find hexapod)/config/controller_configuration.yaml</parameters>
            </plugin>
        </gazebo>
       
    </xacro:macro>
</robot>
```
launch file:
```
import os
from launch_ros.actions import Node
from launch import LaunchDescription
from launch.substitutions import Command
from launch.actions import ExecuteProcess
from ament_index_python.packages import get_package_share_directory

def generate_launch_description():

    robot_model = 'monoped'

    xacro_file = get_package_share_directory('hexapod') + '/models' + '/monoped' +'/xacro/' + robot_model + '.urdf.xacro'

    # Robot State Publisher
    robot_state_publisher = Node(package='robot_state_publisher',
                                 executable='robot_state_publisher',
                                 name='robot_state_publisher',
                                 output='both',
                                 parameters=[{'robot_description': Command(['xacro', ' ', xacro_file])
                                              }])

    # Spawn the robot in Gazebo
    spawn_entity_robot = Node(package='gazebo_ros',
                              executable='spawn_entity.py',
                              arguments=['-entity', 'my_doosan_robot', '-topic', 'robot_description'],
                              output='screen')

    # Start Gazebo with my empty world
    world_file_name = 'my_empty_world.world'
    world = os.path.join(get_package_share_directory('hexapod'), 'worlds', world_file_name)
    gazebo_node = ExecuteProcess(cmd=['gazebo', '--verbose', world, '-s', 'libgazebo_ros_factory.so'], output='screen')

    return LaunchDescription([robot_state_publisher, spawn_entity_robot, gazebo_node])

```

control config.yaml file:
```
controller_manager:
  ros__parameters:
    update_rate: 100

    joint_trajectory_controller:
      type: joint_trajectory_controller/JointTrajectoryController
      
    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

joint_trajectory_controller:
  ros__parameters:

    joints:
      - haa_joint_6
      - hfe_joint_6
      - kfe_joint_6

    interface_name: position
    
    command_interfaces:
      - position

    state_interfaces:
      - position
      - velocity

    state_publish_rate: 200.0 # Defaults to 50
    action_monitor_rate: 20.0 # Defaults to 20

    allow_partial_joints_goal: false # Defaults to false
    open_loop_control: true
    allow_integration_in_goal_trajectories: true
    
    constraints:
      stopped_velocity_tolerance: 0.01 # Defaults to 0.01
      goal_time: 0.0 # Defaults to 0.0 (start immediately)
```
when i launching start_world.launch 

[image]

leg collision doesn't work correctly.

but i delete "hfe_joint_6" from gazebo_control.xacro 
[image]

but now i doesn't control joint.

And one more thing. 
Where should I write the pid parameters in the configuration file? In the ROS1 we should write the config file simply but i can't find any info about writing pid parameters in ROS2.
Thank you in advance for your feedback.

**Environment**
 - Ubuntu 22.04
 - Foxy

destogl, 2023-02-09:
I think you have the following issues:

1. do you have a controller for your leg? It appears that gravity just does its thing.
2. To add PID – you can add this using PID controller available on rolling.

@Zarqu0n I am not sure if this is still working properly on Foxy.

bmagyar, 2023-03-04:
Closing due to inactivity, feel free to reopen

closed 2023-03-04 by bmagyar