ros-controls/ros2_control #885: Mocking: map command interface to state interface of different types

tonynajjar, 2023-01-05:
**Is your feature request related to a problem? Please describe.**

I have a prismatic joint with one velocity command interface and one position command interface. In real world this translates to one linear actuator + a distance sensor tracking the tip.
I was able to create a controller for it. My problem is in running this controller in "simple simulation" with the GenericSystem.
Mirroring the command and state interfaces of the same type does not make sense here, what I need is for the velocity commands to be integrated and mapped to the position state interface - this would represent an idealistic simulation.

Do you think this is something that can be supported by the ros2_control or should I create my own hardware interface that does this?

destogl, 2023-01-19:
> I have a prismatic joint with one velocity command interface and one position command interface.

Do you mean “velocity command interface and one position **state** interface”?

tonynajjar, 2023-01-19:
Definitely yes, edited. Thanks

destogl, 2023-01-19:
@tonynajjar please check my edited post

tonynajjar, 2023-01-19:
I see. I still do wonder if it's a good idea to open up this door; the question if we implement this is where does it stop? What about deriving velocity from position (in the case of position command interface and velocity state interface)? Throw in acceleration in the mix and you have more combinations.

>Do you think this is something that can be supported by the GenericSystem or should I create my own hardware interface that does this?

That's why I was asking this question. Perhaps the GenericSystem should stay as is but have an option to extend it?

destogl, 2023-02-09:
I would personally like to have this. I had it already multiple times when this would be very handy. Implementing this we are covering 90+ % use-cases, from my experience having acceleration commanding interface is not so common – but we can know that first when we try it.

So, yes I would go down this path. I am pretty sure we will not get many questions about doing other crazy things there.

christophfroehlich, 2023-04-11:
I too tripped over this missing feature just now ^^ (velocity command -> integration to position state)

If no one is working on this right now, I'll have a look on the work from @livanov93 and make a PR for rolling. But maybe I just cherry pick the integration-part first.

tonynajjar, 2023-10-04:
Does https://github.com/ros-controls/ros2_control/pull/1028 close this ticket or is there something missing?

christophfroehlich, 2023-10-09:
I tested the same configuration as you were describing in the beginning with the latest feature -> I think we can close this as fixed!

closed 2023-10-09 by christophfroehlich

tonynajjar, 2023-11-09:
Following up on this, you have one more successful test from my side :+1: