ros-controls/ros2_control #894: controller_manager node is not able to check the existence of realtime kernel for Ubuntu 22.04

MorphyCao, 2023-01-11:
**Describe the bug**
controller_manager node is not able to check the existence of realtime kernel for Ubuntu 22.04

**To Reproduce**
Steps to reproduce the behavior:
1. Enable realtime service in Ubuntu 22.04
2. Run control demos

**Expected behavior**
controller_manager logged: RT kernel is recommended for better performance

**Screenshots**
[image]

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Version: humble

**Additional context**
No realtime file exist in /sys/kernel for Ubuntu 22.04 even after enabling the realtime-kernel , however realtimes_tools only check this file to confirm the realtime kernel's existence

bmagyar, 2023-01-16:
@AndyZe you didn't test the detection with the stock RT patch right?

AndyZe, 2023-01-16:
Hmm, I tested this code in ROS1/Ubuntu 18 but not ROS2/Ubuntu 22. It worked in Ubuntu 18.

I believe@tylerjw has used it recently with Ubuntu 22 though. He might have ideas.

tylerjw, 2023-01-16:
I did test this and it worked when I tested it. I don't understand what is wrong.

MorphyCao, 2023-01-17:
Hi, let me try to explain the situation.
1. Only self-built Preempt RT kernel makes sure of the existence of realtime file (/sys/kernel/realtime). Our controller_manager works fine under this scenario.
2. Ubuntu 22.04 now has a native Preempt RT feature (no need user to build their own RT kernel). See this link: https://ubuntu.com/blog/real-time-ubuntu-released.
3. When Preempt RT service is enabled using method described in 2. The file named "realtime" does not exist under /sys/kernel folder.
4. But our controller_manager only checks the file named "realtime" under /sys/kernel folder to judge if the computer runs on a RT kernel.

So I guess our controller_manger should not only check the file named "realtime" but also implement more checking logics for Ubuntu 22.04's native Preempt RT feature

MorphyCao, 2023-01-17:
[image]
I actually modified ros2_control_node.cpp as in screen shot, see if this helps

destogl, 2023-01-19:
@MorphyCao did this help?

Is there any other way to check if there is realtime kernel used / activated?

MorphyCao, 2023-01-30:
> @MorphyCao did this help?
> 
> Is there any other way to check if there is real-time kernel used / activated?
> 
> I would guess there should be some file about that, since that is often how Linux solves things – there is somewhere some file containing a flag or being a flag itself. If this is possible, this would be more preferred solution over cli calls.

I tried using overlay and it worked. Can be a temporary solution for Ubuntu 22.04's native realtime feature.
I totally agree with you and I guess we can try to find a more elegant way to solve this instead of using cli calls.

MorphyCao, 2023-02-07:
The file named "realtime" does exist under /sys/kernel folder now. No need to change the source code to use real-time capability under Ubuntu 22.04

closed 2023-02-07 by MorphyCao

tylerjw, 2023-02-07:
@MorphyCao cool, thank you for commenting here to let us know.

JellyfishKnight, 2023-10-10:
> The file named "realtime" does exist under /sys/kernel folder now. No need to change the source code to use real-time capability under Ubuntu 22.04

@MorphyCao I meet the same situation which you have met before. And I can't find the "realtime" file under /sys/kernel folder. I would appreciate it if you can tell how did you solve this.