ros-controls/ros2_control #897: Example for usage of transmission

firesurfer, 2023-01-12:
Hi everyone, I asked this question first in the `ros2_control_demos` (https://github.com/ros-controls/ros2_control_demos/issues/216) and it was suggested to ask the question in this repository.

I am looking for an example how to actually use a transmission element in the hardware_interface. 

In my xacro file I have a transmission specified like this 

```
<transmission name="mytransmission>
          <plugin>transmission_interface/SimpleTransmission</plugin>
          <joint name="${name}" role="${prefix}${name}_actuator">
                <mechanical_reduction>5.0</mechanical_reduction>
          </joint>
</transmission>
```

When writing a hardware interface i can access the via the `hardware_interface::HardwareInfo &system_info` passed in the on_init method.

Unfortunately I couldn't find any examples for best practices on how to actually use the transmission in the hardware interface.

destogl, 2023-01-13:
@bmagyar that's your experteese :D

firesurfer, 2023-01-25:
See https://github.com/ros-controls/ros2_control_demos/pull/226 for an example

closed 2023-01-25 by firesurfer