ros-controls/ros2_control #908: [Humble backports] Should we backport API change gracefully?

destogl, 2023-01-19:
I would like to merge the API change of adding time and period (#846) on both methods in chainable controllers into humble to, to make the transition simpler and more graceful. Nevertheless, we have to make an ABI change for it. The impact should be little since no many people are using chainable controllers (since there is also no clear documentation about it – shame on me…)

The idea would be the following:

1. Add new method with parameters as virtual.
2. Make default implementation that calls the method without parameters and notifies user about  its deprecation.
3. When user overrides the new method, there is also no deprecation notice anymore.

What do you think?

saikishor, 2023-04-14:
Hello!

yes, I think it would be interesting to have this in humble, it would be really helpful to have it.

Best Regards,
Sai

bmagyar, 2024-11-20:
Timed out.

closed 2024-11-20 by bmagyar