ros-controls/ros2_control #930: Using topics in hardware interface (the method)

AmmarAlbakri, 2023-02-02:
I have a hardware controller that is written in Python and uses topics for command and state values communication, currently, I am unable to convert it to c++ code so I have to connect the hardware interface (c++) with my controller (Python) using topic.

I have successfully used publisher in the write() function to send the command out, but the problem lies within the subscriber and the callback binding.

I can create a subscriber and bind the callback to it but the callback never gets executed whenever messages are sent ?? maybe because of a thread problem??

below is an example of my hardware interface, any help getting the subscriber running is appreciated.

Note: someone already did it before but doesn't explain how [here](https://answers.ros.org/question/403073/ros2-subscriber-in-component/).

## Header file of hardware_interface:
```
#include "hardware_interface/base_interface.hpp"
#include "hardware_interface/handle.hpp"
#include "hardware_interface/hardware_info.hpp"
#include "hardware_interface/system_interface.hpp"
#include "hardware_interface/types/hardware_interface_return_values.hpp"
#include "hardware_interface/types/hardware_interface_status_values.hpp"
#include "rclcpp/macros.hpp"
#include "my_hardware_demo/visibility_control.h"

#include "rclcpp/rclcpp.hpp"
#include "std_msgs/msg/string.hpp"
#include "std_msgs/msg/float32.hpp"

using namespace std::chrono_literals;

namespace my_hardware_demo
{
  class DiffDriveSystemHardware
      : public hardware_interface::BaseInterface<hardware_interface::SystemInterface>,
        public rclcpp::Node
  {
  public:
    DiffDriveSystemHardware();

    RCLCPP_SHARED_PTR_DEFINITIONS(DiffDriveSystemHardware)

    ROS2_CONTROL_DEMO_HARDWARE_PUBLIC
    hardware_interface::return_type configure(const hardware_interface::HardwareInfo &info) override;

    ROS2_CONTROL_DEMO_HARDWARE_PUBLIC
    std::vector<hardware_interface::StateInterface> export_state_interfaces() override;

    ROS2_CONTROL_DEMO_HARDWARE_PUBLIC
    std::vector<hardware_interface::CommandInterface> export_command_interfaces() override;

    ROS2_CONTROL_DEMO_HARDWARE_PUBLIC
    hardware_interface::return_type start() override;

    ROS2_CONTROL_DEMO_HARDWARE_PUBLIC
    hardware_interface::return_type stop() override;

    ROS2_CONTROL_DEMO_HARDWARE_PUBLIC
    hardware_interface::return_type read() override;

    ROS2_CONTROL_DEMO_HARDWARE_PUBLIC
    hardware_interface::return_type write() override;

  private:
    // Parameters for the DiffBot simulation
    double wheel_seperation_;
    double wheel_radius_;

    void topic_callback_left(const std_msgs::msg::Float32::SharedPtr);
    void topic_callback_right(const std_msgs::msg::Float32::SharedPtr);

    rclcpp::Publisher<std_msgs::msg::Float32>::SharedPtr publisher_[2];
    rclcpp::Subscription<std_msgs::msg::Float32>::SharedPtr subscription_[2];
    rclcpp::executors::SingleThreadedExecutor executor;
    
    // Store the command for the simulated robot
    std::vector<double> hw_commands_;
    std::vector<double> hw_positions_;
    std::vector<double> hw_velocities_;

    // Store the wheeled robot position
    double base_x_, base_y_, base_theta_;
  };

} // namespace ros2_control_demo_hardware

```
## The source file of the hardware_interface (only the constructor):

```
DiffDriveSystemHardware::DiffDriveSystemHardware() : Node("hardware_interface_publisher")
  {

    publisher_[0] = this->create_publisher<std_msgs::msg::Float32>("/right_velocity_command", 10);
    publisher_[1] = this->create_publisher<std_msgs::msg::Float32>("/left_velocity_command", 10);

    auto rightCB = [this](const std_msgs::msg::Float32::SharedPtr state) -> void
    { RCLCPP_INFO(rclcpp::get_logger("DiffDriveSystemHardware"), "Right Callback %f", state->data); };

    subscription_[0] = this->create_subscription<std_msgs::msg::Float32>(
        "/right_position_state", 0, rightCB);

    RCLCPP_INFO(rclcpp::get_logger("DiffDriveSystemHardware"), "Added Right Callback");

    auto leftCB = [this](const std_msgs::msg::Float32::SharedPtr state) -> void
    { RCLCPP_INFO(rclcpp::get_logger("DiffDriveSystemHardware"), "Left Callback", state); };

    subscription_[1] = this->create_subscription<std_msgs::msg::Float32>(
        "/left_position_state", 0, leftCB);

    RCLCPP_INFO(rclcpp::get_logger("DiffDriveSystemHardware"), "Added Left Callback");
  }
```

destogl, 2023-02-09:
This is rather a question then enhancement :)

Your issue here is that you never call “spin” or “spin_once” (probably better) on the executor, at least not in the provided code. Therefore, you never check for the callback from ROS.

If this is just demo code call "spin_once" in read or write, if this is serious usage, then use multi-threading and call spin in a separate thread and use real-time buffers to sync the data between the threads. Similar concept as we are doing in controllers.

I think I don't have to tell you that this is highly discouraged usage :) And it is expected that you end up having timing issues.

You are welcome to reopen this is the answer doesn't fixes this for you.

destogl, 2023-02-09:
Now there is an answer in the referenced ROS answers thread.

closed 2023-02-09 by destogl