ros-controls/ros2_control #943: "on_deactivate()" seems not called when setting the controller to be "inactive"

xibeisiber, 2023-02-20:
Hi,
I tried `diffbot.launch.py` and `rrbot_system_position_only.launch.py`. The printing information in `on_activate()` is printed out in console, but when I "ros2 control set_controller_state forward_position_controller inactive" and "ros2 control set_controller_state diffbot_base_controller inactive", the printing information in "on_deactivate()" is not printed in console.

Another question is: will `on_deactivate()` be automatically called when I terminate the startup node (press ctrl+c in` diffbot.launch.py `)?

Thanks!
Jia

christophfroehlich, 2023-03-02:
repost from https://github.com/ros-controls/ros2_control_demos/issues/239

I'll try to reproduce your issue.

destogl, 2023-03-03:
since repost, closing

closed 2023-03-03 by destogl